RoganDawes/P4wnP1 #92: Error if i execute the command "shell"

Anonyvius, 2017-09-05:
Hey. I connected me with the command Firestage1. Then i executet the command "shell" but i got this error:

[image]

how to fix it?

Anonyvius, 2017-09-05:
or in text output:

`P4wnP1 shell (client connected) > shell
Process with ID 1360 created
Trying to interact with process ID 1360 ...
Microsoft Windows [Version 10.0.15063]
(c) 2017 Microsoft Corporation. Alle Rechte vorbehalten.

C:\Users\User>Exception in thread P4wnP1 Server Output Loop:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "P4wnP1.py", line 240, in __output_handler
    method = pending_methods[method_id]
KeyError: 4L

Client received terminate!
P4wnP1 shell (client connected) >
Target disconnected
P4wnP1 shell (client not connected) >
Client disconnected, stop interacting
P4wnP1 shell (client not connected) >
Client didn't respond for 1 seconds.
P4wnP1 shell (client not connected) > ServerChannel: channel ID 1 onClose called
ServerChannel: channel ID 2 onClose called
ServerChannel: channel ID 3 onClose called`

mame82, 2017-09-05:
This error has been reported here
https://github.com/mame82/P4wnP1/issues/75

Are you able to bring up a reproducable test case?

mame82, 2017-09-05:
Hopefully fixed by this commit https://github.com/mame82/P4wnP1/commit/9ec29777ee6c0b63a4d2cd16007236ed26e56d90

Anonyvius, 2017-09-05:
Not realy sometimes it just happens.

Anonyvius, 2017-09-05:
But i am triening to figure it out

mame82, 2017-09-05:
Shouldn't happen anymore, if you pulled the latest commit

closed 2017-09-09 by mame82