RoganDawes/P4wnP1 #95: FireStage1 with german keyboard layout does not work

Dexxtra, 2017-09-06:
While the english one does work like a charm, the german keyboard language does not. It duplicates some keys which causes for example the word Runtime to end up RRRRRRRRRRRRRRRRRuntime.

mame82, 2017-09-06:
Hmmm ... one more issue since merging in an external duckencoder.py code overhaul.

Can you give steps to reproduce (I haven't got this issue) ?

Are you working with latest master branch commit ?

What is the result of `SendKeys Runtime` with German Keyboard layout.

What OS is the test target running on?

Dexxtra, 2017-09-06:
Latest master branch, Target OS is Windows 10 Creators update. When i change the keyboard layout on both Windows and in P4wnp1 it works. I can give you the result of Sendkeys Runtime later as I am at work now.

mame82, 2017-09-07:
Nice!

closed 2017-09-07 by mame82