RoganDawes/P4wnP1 #96: payload overwrites important settings in setup.cfg

Swiftb0y, 2017-09-07:
I dont know if this is supposed to be like this but most/every payload overwrites the lang variable which breaks payloads that dont have a cli like the hid_backdoor.

The USB_VID and USB_PID get also overwritten every time which is only necessary on payloads like the Responder.

The hid_backdoor_remote also overwrites the WIFI_AP, AUTOSSH and WIFI_CLIENT rather than checking if the variables are set and sets them based on that instead of just stupidly overwriting them.

Would you fix that or should I work on a PR?

mame82, 2017-09-07:
Settings of the payload have priority over settings in setup.cfg.

This is the intended behaviour, as the payloads are more specific.
If you want to use settings from setup.cfg remove the respective option from the payload.

For example, removing the `lang` option from a payload would force usage of `lang` from setup.cfg.

OS fongerprinring is possible with 3rd party tools (f.e. Responder, nmap ...)

Swiftb0y, 2017-09-07:
> OS fongerprinring is possible with 3rd party tools (f.e. Responder, nmap ...)

I know but is there a way to just identify if the system is running windows/*nix?
This would allow payloads to decide if they should execute a powershell- or perl stager.
Firing up responder or nmap and scraping the output seems a little slow to me.

> Settings of the payload have priority over settings in setup.cfg.

Im also aware of that and it makes sence that the payload only activates the hardware features it needs but it shouldn't override the AP and Client settings in my opinion.
Just checking if they are empty with something like `if [ -z "$VAR" ];` (which returns true if unset or empty) and setting it based on that would be the better approach, since the user would just have to tamper with the setup.cfg instead of the individual payloads.

And the lang override doesnt make any sence at all since it gets set in every single payload.

mame82, 2017-09-07:
According OS fingerprint:
Research exists on OS detection on USB basis, but this couldn't be done on top of Raspbian (has to be done on low level, do it is out of scope for me).

According setup options:
I'll keep this behavior. This would allow payloads to behave differently. If you want to use global WIFI options delete them from the redpective payloads, set the ones in setup.cfg accordingly and you're good to go.

Setup.cfg is global, as long as options aren't defined more specific in payload. So I don't see the need to change this behavior for single options.

Swiftb0y, 2017-09-07:
> I'll keep this behavior.

I understand. I was just saying that this behavior might be confusing for people that dont know the P4wnP1 internals.

When USB0 and USB1 are "emulating"  RNDIS and CDC ECM adapters, and the other one is deactivated once a successful connection occured the Pi should know if RNDIS or CDC ECM is used and set a Variable like IS_WINDOWS accordingly so other script can use it. Shouldnt that be possible or am I understanding some fundamentally wrong?

mame82, 2017-09-07:
Addition:
Every combination of USB gadgets (UMS, HID keyboard, RAW_HID, CDC ECM, RNDIS) should use a dedicated VID + PID combination. Windows for example caches drivers based on this, a known device isn't reenumerated. Changing the USB gadgets without changing PID or VID could lead to failing driver installation (for a gadget which wasn't present on first install).

That's why every demo payload uses different PID+VID, when USB setup differs

Swiftb0y, 2017-09-07:
> That's why every demo payload uses different PID+VID, when USB setup differs

but then whats the purpose for it to be in the setup.cfg?

mame82, 2017-09-07:
> When USB0 and USB1 are "emulating" RNDIS and CDC ECM adapters, and the other one is deactivated once a successful connection occured the Pi should know if RNDIS or CDC ECM is used and set a Variable [...]

Not reliable, many Linux distros support RNDIS, some only ECM (as MacOS).
A payload isn't forced to use USB networking, which would be another problem.

If YSB networking is used `onNetworkUp` provides a variable holding the active interface, called `$active_interface`

mame82, 2017-09-07:
> but then whats the purpose for it to be in the setup.cfg?

To have a default, in case it isn't defined in a custom payload. In fact a payload could be an empty file, althoufh this doesn't make sense

Swiftb0y, 2017-09-07:
Ok. then nevermind the question. The only thing I would change the lang stuff since it doesnt make any sence.

mame82, 2017-09-07:
These questions are important, as I barely have time to write docs.

According `lang` option. I work in multinational environments from time to time, having multi language payloads ready could be helpful. But you're right ... the more common case is that one wants to change the language of all keyboard payloads at once

Swiftb0y, 2017-09-07:
ok. about the docs. I could try to write some since I consider myself to be somewhat familiar with the bash payload system and the payloads themselves. If that would help the project...

closed 2017-09-09 by mame82