RoganDawes/P4wnP1 #97: USB Mass Storage

Hugand, 2017-09-07:
When the USB Mass Storage is enabled in a payload all of the files that are collected from a system are accessible in the folder "collected" right? The thing is, when I ssh into the pi to see the files, they simply aren't there.

mame82, 2017-09-07:
No, the USB storage is backed by a 128 MB vfat image.
This image is stored in `$wdir/USB_STORAGE/image.bin` and contains the files stored to UMS.
($wdir is the working dir of P4wnP1)

The collected folder, should be used by payloads which want to store data to P4wnP1's internel Linux file system.

The UMS image couldn't be loop mounted to Linux, while used as flashdrive (produces inconsistent data unfotunately )

Hugand, 2017-09-07:
Oh ok, so what should I do to get the files from that image.bin? (sorry if this is an ignorant question but I recently got into Linux and I'm still learning)

PoSHMagiC0de, 2017-09-08:
@Hugand 
I have not tested this myself.  I have been running the hid_backdoor through the ringer but here is what I can guess.
Easiest way would be to plug the Pi into your machine with the USB storage payload selected (one that you do not have some autorun stuff on) and then you can access it like a USB drive to get your stuff off.

Longer way, copy the bin file to your machine (most likely a *nix machine).
Follow directions to setup a loopback mount for a disk image file locate at links below.

https://www.linuxquestions.org/questions/linux-general-1/how-to-mount-img-file-882386/

https://www.youtube.com/watch?v=KchAIeQkBxE

Maybe in windows you could use something like OSFMount.
https://www.osforensics.com/tools/mount-disk-images.html

Hugand, 2017-09-08:
Thanks a lot dude!!

closed 2017-09-09 by mame82