RoganDawes/P4wnP1 #102: [Not an Issue] Best way to update?

jcstill, 2017-09-08:
Hey @mame82, what is the best way to update the P4wnP1 software?

Reason I ask is bc I don't have a USB - LAN adapter or mini hdmi cable so I have to install the image to the micro sd and then use my Pi B+ to run `install.sh`. Once that runs and I am able to connect to the wifi, and just ssh. 

I just want to know is there a better way to update rather than having to install everything from scratch? 

I also was wondering bc installing from scratch is a pretty lengthy process for small updates.
Thanks
J

Swiftb0y, 2017-09-09:
you can connect the pi to an existing wpa2 network by using the wpa_supplicant.

1. ssh into the pi from your PC (not via P4wnP1 AP)
2. `cd /etc/wpa_supplicant`
3. `wpa_passphrase <your ESSID> <your passphrase> >> wpa_supplicant.conf`
4. `wpa_supplicant -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf`
5. `sudo dhclient wlan0`

Your P4wnP1 should now be connected to your wifi. You can check if you have gotten an IP-address by using `ifconfig` and internet connectivity by `ping http://scanme.nmap.org/` or `ping 8.8.8.8`.
Hope this helps.

mame82, 2017-09-09:
On the latest release, the easiest way to get Internet access id to use the `wifi_connect` payload. (The `WIFI_CLIENT` options have to be adjusted in order to connect to an existing WiFi with internet access).

P4wnP1 is accessible from the WiFi via SSH, the IP depends on the DHCP lease offered by existing WiFi.

Now that you're able to SSH in to the Internet connected Pi, the update process depends on the changes between your version and the current commit in master branch:

1) If only files of P4wnP1 have changed (install.sh unchanged; no submodules like Duckencoder.py, Responder or JtR changed), it should be enough to pull the new files by running `git pull` from P4wnP1's main directory.

2) In case submodules have been modified, they have to be update, too, in addition to step 1). This could be done with `git submodule update --recursive --remote`

3) In case the installer changed (which mostly is due to added software packages, like AutoSSH in the latest release), the installer has to be ran again with `install.sh`.
This should take less time than the first install, as most of the requiremebts are already met.

So to be sure everything is installed as intended, all 3 steps mentioned above should be done.

The easiest way to combine all steps into one:
delete the `P4wnP1` directort and rerun `git clone --recursive https:://github.com/mame82/P4wnP1 && cd P4wnP1 && ./install.sh`.

It wouldn't be a good idea to include an updater script in this project stage, as changes are to frequently and there exist to many corner cases.

So my recommendation is to always start with a clean re-install, as long as major changes are introduced to the repo frequently

PoSHMagiC0de, 2017-09-09:
To add.  If you looking to speed up the process, make a new base image.  With me I downloaded the lite version of raspbian stretch and imaged it to a 2GB microSD (smallest you can use).  I preprepped it for headless usb_ethernet like in all the tutorials out there with ssh enabled.  I also prefilled the interfaces file with a config similar to what P4wnP1 would give but put it on a class C subnet for simplicity and the gateway pointing to what my PC IP will be for the usb interface.
I put it in the pi, boot it up.  When I am able to and the interface is stable I add a static IP to my usb interface the same as my pi is pointing to for gateway.  I use linux so I enable ip forwarding and set my postrouting nat to masquerade (killing all my other preset rules first).  I ssh into my pi, apt update,upgrade,dist-upgrade and autoremove the system and then shutdown.  I then make an image of that image from the SDCard and use that as my new base.  Now when I image it to a bigger SD card I have to remember to to a raspi-config to resize my filesystem.  I tried adding back in that cmdline argument in cmdline.txt in the boot partition that is suppose to resize but it seems to resize the partition but not the filesystem so I still have it in there but have to still run raspi-config to fix the filesystem.
Anyway, that will give you a base to go to, to start new and quickly each time.

I normally just do a git pull from the P4wnP1 folder for updates depending on what has been changed since last pull so I do track the changes mame makes.  If it is a major change to like the kernel modules or what not, I redo pi.  If it is just to his python modules like for the fix he was doing for the method_id, I just do a pull but I backup any files I made changes to, which is mostly the setup.cfg.
Note: If mame made changes to the setup.cfg and you did, the pull will fail so you will need to rename the setup.cfg if that happens to get the pull.  Yeah, if in doubt just redo.

jonas18z, 2017-09-20:
”The easiest way to combine all steps into one:
delete the P4wnP1 directort and rerun git clone --recursive https:://github.com/mame82/P4wnP1 && cd P4wnP1 && ./install.sh.”

Is it possible in some way to keep, user settings, my duckyscripts Etc when you update?

mame82, 2017-09-20:
DuckyScripts - backup using SCP pr WinSCP.

Settings - I don't recommend to backup, last commits introduced a bunch of new settings (mouse, bluetooth, autossh, WiFi client etc.) 

The latest commits need a re install, due to changes to udev + systemd + package upgrade.

Written payloads should still be usable (only one parameter was removed `USB_RAWHID_FULLDUPLEX`)

closed 2017-09-26 by mame82