RoganDawes/P4wnP1 #103: How do I add duckyscripts

YepJustAGamer, 2017-09-09:
I have been trying to figure this out. I have looked on the sd card but I can't find it?

3FF-53C-Member-DalmAsian, 2017-09-09:
you must be in a Linux computer to see the partition on the SD card that has all the system folders and it will be under home folder then the p4wnp1 folder then you'll see the duckyscript folder.

3FF-53C-Member-DalmAsian, 2017-09-09:
@YepJustAGamer you must be in a Linux computer to see the partition on the SD card that has all the system folders and it will be under home folder then the p4wnp1 folder then you'll see the duckyscript folder.

Hugand, 2017-09-09:
and how can I run a ducky script after encoding it with the duckyencoder?

3FF-53C-Member-DalmAsian, 2017-09-09:
useing the backdoor payload and  using the SendDuckyScript command

Hugand, 2017-09-09:
ok

mame82, 2017-09-09:
This was already answered in other issues.

closed 2017-09-09 by mame82