RoganDawes/P4wnP1 #104: USB interfaces stopped working

Swiftb0y, 2017-09-09:
I just bricked my PI so I had to reinstall raspbian and decided to upgrade to stretch.
Fortunately I am still able to connect to it while in wifi-client or AP mode.
I am using the latest raspbian stretch with up-to-date P4wnP1 sources and a otherwise fresh install. My host is a Laptop with Kali Linux.
The Issue is most likely somewhere in `boot/init_usb_ethernet.sh line:53 since thats where the dots (seen below) come from`
These are the last lines of `sudo journalctl -u P4wnP1.service`
`Sep 09 18:00:27 MAME82-P4WNP1 bash[186]: USB OTG off, going on with P4wnP1 boot
Sep 09 18:00:27 MAME82-P4WNP1 bash[186]: P4wnP1: ... USB gadget initialized
Sep 09 18:00:27 MAME82-P4WNP1 bash[186]: P4wnP1: Initializing Ethernet over USB...
Sep 09 18:00:27 MAME82-P4WNP1 systemd[1]: Started P4wnP1 Startup Service.
Sep 09 18:00:27 MAME82-P4WNP1 bash[186]: CDC ECM and RNDIS active. Check which interface has to be used via Link detection
Sep 09 18:00:30 MAME82-P4WNP1 dhclient[337]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 8
Sep 09 18:00:30 MAME82-P4WNP1 dhclient[337]: DHCPREQUEST of 192.168.178.50 on wlan0 to 255.255.255.255 port 67
Sep 09 18:00:30 MAME82-P4WNP1 dhclient[337]: DHCPOFFER of 192.168.178.50 from 192.168.178.1
Sep 09 18:00:30 MAME82-P4WNP1 dhclient[337]: DHCPACK of 192.168.178.50 from 192.168.178.1
Sep 09 18:00:31 MAME82-P4WNP1 dhclient[337]: bound to 192.168.178.50 -- renewal in 365542 seconds.
Sep 09 18:18:41 MAME82-P4WNP1 bash[186]: ..................................................................................................................... (goes on for pretty much forever)
~`

mame82, 2017-09-09:
The dot is printed here https://github.com/mame82/P4wnP1/blob/master/boot/init_usb_ethernet.sh#L56

The routine doing this, polls both of the internal interface (RNDIS + CDC ECM) for link from the Pi's end.
On a real etherney interface this would happen, as soon as a cable is connected.

On ECM and RNDIS this should happen, as soon as the driver for the corresponding host interface is ready and the **interface is up**.

As you're using kali, both interfaces should be detected from this end (an RNDIS driver is present, too).

It has been a while, since I tested against Kali and I'm not able to do it right now.

But my best guess is, that the interfaces are down on Kali's end or there's an issue with the drivers.

So I could only give hints, but no solution.

On Kali, check if USB interfaces are present (named `usb*` depending on your config).
This could be done with `ifconfig -a`
If you spot two interfaces, enable both with `ifconfig usb1 up`, where the `1` has to be replaced with the correct number.

Now one of the interfaces should receive link, on P4wnP1's end.

Note: P4wnp1 tries to disable the internal RNDIS interface and use ECM if both are present. The problem is that this decision is taken very quickly, while you're enabling the Kali interfaces relativly slow. Thus I can't tell for sure if you hit the ECM interface correctly. `dmesg` on kali should come to help to identify the correct device naming (watch out for USB message).
I recommend using CDC ECM on Linux and to setup the interface for auto configuration.

Swiftb0y, 2017-09-09:
I AM SO FUCKING STUPID!!!!
I THOUGHT I CHECKED EVERYTHING BUT I JUST CONNECTED THE PI VIA THE POWER ONLY USB PORT!!!
sorry to bother you :-(

closed 2017-09-09 by Swiftb0y