RoganDawes/P4wnP1 #106: How hard would it be to add enterprise connections to client mode?

walletdude, 2017-09-10:
I love the client mode feature of the project, but I wish it supported modes other than WPA2.  How hard would it be to support enterprise networks for client mode?

Swiftb0y, 2017-09-11:
It depends. The wpa config get generated [here](https://github.com/mame82/P4wnP1/blob/master/boot/init_wifi.sh#L148) and [here](https://github.com/mame82/P4wnP1/blob/master/boot/init_wifi.sh#L189). As you can see, a temporary config gets created and appended to the existing networks from /etc/wpa_supplicant/wpa_supplicant.conf. If you know how to manually add an enterprise wifi there, you can just add it normally and remove the check [here](https://github.com/mame82/P4wnP1/blob/master/boot/init_wifi.sh#L217). This would be the easiest temporary hack I could think of, but you might brick your existing installation if you are not careful.
It should not be to difficult to implement a genuine solution, but other things have more priority right now.

mame82, 2017-09-18:
Not planned at the moment (PR is an option, but code changes frequently)

DeadPackets, 2017-09-28:
I might be able to work on this, what is exactly needed to be changed? I'm having trouble understanding what the scripts do exactly.

Swiftb0y, 2017-09-28:
As already stated, I suggest you read the wiki. The P4wnP1 core is a systemd service that calls boot/boot_P4wnP1 which is a bash script that includes the modules based on the settings. The wifi gets setup in boot/init_wifi.sh but what for an ok from mame before you modify the core.