RoganDawes/P4wnP1 #110: hid_keyboard.txt nor hid_backdoor.txt is working

gonebad1222, 2017-09-16:


gonebad1222, 2017-09-16:
Also, yes I set the keyboard to the correct instance "US". I just wanted to make sure I gave enough information as possible and realized that I had not mentioned that.

jcstill, 2017-09-16:
What do you mean? How is it not working? Are you getting error messages?

gonebad1222, 2017-09-17:
N
Nothing happens I don't see any error messages. Within Windows. When the HDMI cable is plugged in I see no error messages from the Pi zero (w) either.

gonebad1222, 2017-09-17:
Can I put P4wnP1 in debug mode so I can see more output?

jcstill, 2017-09-18:
What happens when you boot the pi? 
Have you followed the install.md exactly? 
Are you SSHing? 
Have you enabled the backdoor payload? (you need to disable all the other ones)
Run `sudo python /home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py`
Do you get anything?

There is no debug mode to my knowledge
And you all of the error messages (if any) would be on the terminal. Whether it be SSH or HDMI.
Watch seytonic's video on the P4wnP1. It is a major help in setting it up.

jcstill, 2017-09-18:
Also, please excuse me if I am coming off as rude. I just didn't know how else to put those questions.

Look through the other issues (as well as the closed ones). They may give you some insight.

Swiftb0y, 2017-09-18:
A `sudo journalctl -u P4wnP1.service` will show you the debugging output P4wnP1 created. We can probably help you if you post that.

gonebad1222, 2017-09-21:
thank you Swiftb0y. Also, yes I have done these things jcstill. I am not a novice programmer. Just very intrigued by this project and once I get a grasp of it(working) I would like to contribute. All of the work put in by mame82 and others is amazing. I hope I can be of use. Also, I am about to start development on a self driving car so I will be caught up with that as well. Will update once i have followed Swiftb0ys advice. By nature I am a C# developer. SO there is a bit of a learning curve. However, I do learn very fast, if I do say so myself. Will Update. Thanks :)

jcstill, 2017-09-21:
Self driving car? That's pretty sick to make one yourself. Especially if you start from scratch. Good luck on that!

gonebad1222, 2017-09-21:
Well with a few libraries on Neural Logic and open source projects on automated cars it makes the work easier... The internet is a wonderful thing my friend.

Swiftb0y, 2017-09-21:
It would be great to have a new contributor. To get you started:
P4wnP1 internals are written in bash,
The backdoor cli is written in Python2
and the backdoor server (running on the target) is written in C# (I believe).
To get you started: [here](http://github.com/mame82/P4wnP1-Wiki) is our still WIP wiki.
And another debugging option is hidden in hidtools/backdoor/P4wnP1.py in the constructor of the main class.

gonebad1222, 2017-09-22:
OK during a re-installation (from kali linux -- this also had to be done with an ethernet to micro usb adapter -- i do not know why trying to figure that out but it is irrellevant to my issue as I was able to get it connected to the internet with a few mods to the setup process written by mame) its a few lines sorry so long.

`Installing` needed python additions...
Requirement already satisfied: pycrypto in /usr/lib/python2.7/dist-packages
Collecting pydispatcher
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python2.7/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/usr/lib/python2.7/dist-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 643, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py", line 315, in increment
    total -= 1
TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'

gonebad1222, 2017-09-22:
Also it is a pi zero wireless with the latest version of raspbian lite download.

mame82, 2017-09-22:
You messed something during install process. The error you're facing is produced by `pip install pydispatcher` . According to the python stack trace, it seems a HTTP GET request to some url fails (maybe Internet connection loss)

The installer is tested against latest Raspbian Stretch light.

I suggest to follow the headless install instructions, which don't require an USB ethernet adapter and are based on Kali Linux, which you are already using.

If you've got a WiFi network running, which provides internet access it could been done even easier, following this instructions:

https://dantheiotman.com/2017/09/15/p4wnp1-the-pi-zero-based-usb-attack-platform/

gonebad1222, 2017-09-22:
My internet has been intermittent. the problem is I do not get a usb0 that comes up in ifconfig.

mame82, 2017-09-22:
Again, as the installer doesn't finish without issues, it doesn't make sense to search for the root cause of a non working RNDIS composite gadget or are you refering to `usb0` on Kali's end ?

gonebad1222, 2017-09-22:
that is correct.

gonebad1222, 2017-09-22:
I do believe the installation was a problem with connectivity through my internet as I have been having problems all last night. I am currently in the middle of downloading and installing currently; will update shortly with the progress.

gonebad1222, 2017-09-22:
update-rc.d: error: cannot find a LSB script for ntp
this was the only error i ran into

mame82, 2017-09-22:
The 'ntp' error is fine, as the installer tries to disable this service, but it only exists on Jessie (not on Stretch). This ha no impact on functionality.

gonebad1222, 2017-09-22:
Awesome I will comment out the networkonly.txt and uncomment the keyboard attack to see if i can get just the Keyboard demo a test run. Will report back after a reboot and trial.

MeantCoder, 2017-09-22:
@gonebad1222 How to installed P4wnP1? if you can't follow INSTALL.md try this : "**git clone --recursive https://github.com/mame82/P4wnP1**" my problem fixed with recursive installation

Swiftb0y, 2017-09-22:
That's correct. The duckencoder, John the Ripper and the Windows Lockpicker are necessary Submodules which get only cloned with the --recursive flag!

MeantCoder, 2017-09-22:
Can you connect HDMI and please send log

gonebad1222, 2017-09-23:
no guys i have it installed... without errors

Swiftb0y, 2017-09-23:
That's great. Could you close the issue then?

gonebad1222, 2017-09-23:
the problem is things happen as soon as I alter the setup.cfg in any way.. such as renaming the wifi access point to something else.. it will show  a "hidden network" or sometimes show the original P4wnP1 network. Now when it is hidden I have to use the accespoint name that I came up with in the setup.cfg with the original P4wnP1 password MaMe82-P4wnP1 (even though I have changed it). I may have a faulty pi. But i have reinstalled with out any errors numerous times. I will even MAIL Mame82 my pi so he can take a look at it... because this is very strange.... Im purchasing like 10 more for various projects here soon so i dont care to mail it.. i just cant figure it out.

I am from the old school botnet days with bufferover run exploits and botnets(rpc,dcom,netbios) in IRC chatrooms that you ran secret IRCDS on something like a universities library i loved hacking when I was younger and I want to contribute as much as I can to this community. When I get a new Pi here in a couple of days I will try again. Mame82 if you would like me to mail my current one to you I will so you can poke around in it yourself if you would like. Just let me know

gonebad1222, 2017-09-23:
Also yes the hidden network feature was set to false and i did it through the shell sudo nano setup.cfg also manually with a keyboard connected to the pi itself.

Swiftb0y, 2017-09-23:
Mhh ok. Just as a last attempt: Did you check if the variables you set get overridden by the payload you chose?

gonebad1222, 2017-09-23:
the original payloads were left intact "as is" never touched anything in there. looked around. never touched.

gonebad1222, 2017-09-23:
they were left straight as they were "out of the box"

Swiftb0y, 2017-09-23:
That's what I mean. Most of them override the settings in setup.cfg! I'm pretty sure that's the issue.

gonebad1222, 2017-09-23:
.... if i overlooked this... im putting my head through a wall...

gonebad1222, 2017-09-23:
i actually did something i usually never do which was assumed they work with one another... hmm i'm taxed tonight will look into in the morning i have 2 more hours of coding to do on my own project and then have to get up three and a half hours later.

gonebad1222, 2017-09-23:
but i have an idea of running a small "Live" version of a simple OS that reads and replaces hashes with a known password hash... restarts logs in... then executes arbitrary code restarts once done and replaces the original one back. would require a couple of minutes around the target machine but it is an idea. let me know what you guys think.

gonebad1222, 2017-09-23:
bc we all know you cant access the hash file wile the windows OS is running.

Swiftb0y, 2017-09-23:
That's an Idea, but if you are talking about Windows, you could just replace sethc.exe with powershell.exe and press shift 5x on startup to input malicious code into a locked machine. And there is also drive encryption...

Swiftb0y, 2017-09-23:
But good idea though. And I would really like to know if you overlooked the payload override.

gonebad1222, 2017-09-23:
true but is the encryption on windows standard?

gonebad1222, 2017-09-23:
I'm not working on it at the moment. Markov decision tree is what i am working on at the moment.

gonebad1222, 2017-09-25:
I think it may be a problem in the hid interface completely i'm starting with the keyboard. running diagnostics give me a few. hopefully someone is up

gonebad1222, 2017-09-25:
I fixed it... i had to un comment this to get it to fire onKeyboardUp
`#function detect_HID_keyboard()
#{
#	echo "Waiting for HID keyboard to be usable..."
#
#	# blocking read of LED status
#	python -c "with open('/dev/hidg0','rb') as f:  print ord(f.read(1))"
#	# fire 'onKeyboardUp' after read has succeeded
#	declare -f onKeyboardUp > /dev/null && onKeyboardUp
#
#}
`
 comes like this out of the box from the git repo

mame82, 2017-09-25:
This issue looks a bit confusing. The code changes highlighted on the former post are misleading and have no real effect, because the `detect_HID_keyboard` function never gets called.

HID keyboard seems to be working according to user feedback. 

So I encourage evertbody to provide reproducable  steps leading to a keyboard error here. If there doesn't exist a general error, I'm going to close this to avoid further confusion

closed 2017-09-26 by mame82

MrUser77, 2023-12-21:
I had the same problem. I could solve it with a change of the cable. Than it workt.