RoganDawes/P4wnP1 #113: Win10_LockPicker john libssl error

pixelwasser, 2017-09-18:
Hello,

at first thank you for this cool Project. But I have a Problem initalizing Win10_LockPicker. During the start I am getting that John could not find libssl libary and then nothing is happening anymore. The hash only includes the current username.

Windows 10 was 15063.608

pixelwasser, 2017-09-18:
Sorry Image was Stretch Lite Image maybe the payload is not working with this Version. I also installed libssl-dev afterwards but no Change.

mame82, 2017-09-18:
Have to check this

jcstill, 2017-09-18:
I am getting similar results with the backdoor payload right after enabling. you have to reboot the pi for changes to setup.cfg to take effect.

pixelwasser, 2017-09-22:
Hello,
today I had a Little time and I installed everything from scratch new and I also compiled John on the Pi Zero W myself and copied the files from run Folder to the P4wnP1 Folder, within the syslog file i found following notifications: 

Sep 22 22:00:06 MAME82-P4WNP1 bash[492]: /tmp/payload_vars: line 31: syntax error: unexpected end of file

Sep 22 22:02:25 MAME82-P4WNP1 bash[190]: Captured the following hashes:
Sep 22 22:02:25 MAME82-P4WNP1 bash[190]: Jan::DESKTOP-HT39ACO: (Hash removed)
Sep 22 22:02:25 MAME82-P4WNP1 bash[190]: .
Sep 22 22:02:25 MAME82-P4WNP1 bash[190]: Starting JtR and trying to crack...
Sep 22 22:02:25 MAME82-P4WNP1 bash[190]: /home/pi/P4wnP1/john/john: error while loading shared libraries: libssl.so.1.0.0: cannot open shared object file: No such file or directory
Sep 22 22:02:25 MAME82-P4WNP1 bash[190]: All given hashes cracked!!!
Sep 22 22:02:26 MAME82-P4WNP1 bash[190]: /home/pi/P4wnP1/john/john: error while loading shared libraries: libssl.so.1.0.0: cannot open shared object file: No such file or directory
Sep 22 22:02:26 MAME82-P4WNP1 bash[190]: /home/pi/P4wnP1/john/john: error while loading shared libraries: libssl.so.1.0.0: cannot open shared object file: No such file or Directory

Hope this helps.

pixelwasser, 2017-09-22:
The Cracked Hash File is empty with Zero Bytes

mame82, 2017-09-23:
I'm able to reproduce the error ... under investigation

pixelwasser, 2017-09-23:
I am wondering why this also with my self compiled Version
Currently I am installing jessie to see if the bug is also on this Version present.

mame82, 2017-09-23:
Should be fixed with latest installer, working on my end now!
Could you test (re install needed) ?

pixelwasser, 2017-09-23:
Yes later the day, I have just checked out Jessie and here everything is working

mame82, 2017-09-23:
Yes, I patched JtR Jumbo source of the old package to work with Jessie before compiling, but other features like bluetooth support are only tested against Stretch.

Recompiling JtR for Stretch didn't involve source patching, so I'll drop Jessie support as soon as possible

pixelwasser, 2017-09-23:
It works now also with Stretch. 

Thanks

closed 2017-09-23 by pixelwasser