RoganDawes/P4wnP1 #115: Hakin9 tutorial doesn't work

MJGames, 2017-09-19:
Hi, the payload Hakin9 tutorila doen't work. Always it waiting for the drive, but the drive is mounted. I use Win7.

mame82, 2017-09-22:
What's the name of the thumb drive ?

mame82, 2017-09-26:
Please reopen if the issue remains

closed 2017-09-26 by mame82

bpartin2009, 2018-02-06:
I see the same issue. Always wait on drive but it is mounted to E: drive as name "HAKIN9"

bpartin2009, 2018-02-06:
Windows 10

bpartin2009, 2018-02-06:
PS C:\Users\1111633> $drive=Get-Volume -FileSystemLabel "HAKIN9" -ErrorAction Stop
Get-Volume : Invalid property
At line:1 char:8
+ $drive=Get-Volume -FileSystemLabel "HAKIN9" -ErrorAction Stop
+        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : MetadataError: (MSFT_Volume:ROOT/Microsoft/...age/MSFT_Volume) [Get-Volume], CimException
    + FullyQualifiedErrorId : HRESULT 0x80041031,Get-Volume

bpartin2009, 2018-02-07:
So apparently I wasn't signed into said computer as sys admin or root user. I for so this to work, sign on as sys admin, use an OTG cable that can transmit data, and I had to add OTG=false on line 116 of /P4wnP1/boot/boot_P4wnP1 file after the hostmode is checked. Still, all in all, really cool project. I will see if I can't create addiontional functionality. "The world is my oyster"