RoganDawes/P4wnP1 #117: Only hid mouse works

99bit, 2017-09-24:
First of all i installed p4wnp1 by wifi adapter as i write script that run ./install.sh after connecting internet 
After 20 min suddenly screen turned black 
Than i tried every payload but only hid_mouse works other payload not works.

gonebad1222, 2017-09-25:
A) what screen turned black? the computer screen or the HDMI output?
B) I am running some diagnostics on the HID itself. Swiftb0y can give some insight aswell. I believe it has to do with some of windows updates that have been coming up.
C) Let us know how much coding experience you have :) 
Thanks!
-g0n3b4d

99bit, 2017-09-25:
yes screen turned black

99bit, 2017-09-25:
HDMI output screen turned black

99bit, 2017-09-25:
i Don't have more experience still studying in school

mame82, 2017-09-25:
The console screen of Pi turns black after some minutes without interaction (there isn't direct interaction when a payload is running). This seems to be some energy saving feature, I haven't done research on how to disable this feature. Suggestions are welcome

99bit, 2017-09-25:
thxx now it works perfect 😉😉

mame82, 2017-09-25:
Nice

closed 2017-09-25 by mame82