RoganDawes/P4wnP1 #118: FIXED keyboard not working

gonebad1222, 2017-09-25:


gonebad1222, 2017-09-25:
/P4wnP1/boot/init_hid_keyboard.sh take the " # " ( or uncomment) out from in front of everything EXCEPT what has " # " in this already (you could copy and paste this directly there.

function detect_HID_keyboard()
{
echo "Waiting for HID keyboard to be usable..."

# blocking read of LED status
python -c "with open('/dev/hidg0','rb') as f:  print ord(f.read(1))"
# fire 'onKeyboardUp' after read has succeeded
declare -f onKeyboardUp > /dev/null && onKeyboardUp
}

mame82, 2017-09-25:
Uncommenting the `detect_HID_keyboard` function should change nothing, as this function doesn't get called on boot.

The code used by this (deprecated) function is used in the boot script
https://github.com/mame82/P4wnP1/blob/master/boot/boot_P4wnP1#L164

So uncommenting it and re-activating the function call would lead to a redundant read from keyboard device, which could block execution till an LED changes.

So could you please give steps to reproduce the isdue description "keyboard not working". I'm not aware of a keyboard issue, nor am I able to reproduce it.

gonebad1222, 2017-09-25:
plain and simple... it was the only way i could get the keyboard working from a fresh install of the get repo --recursive ... soo i don't know

gonebad1222, 2017-09-25:
maybe take a look at the code.. by the way i was in no way trying to be rude.

gonebad1222, 2017-09-25:
Also I followed the instructions EXACTLY to install from kali with no errors

mame82, 2017-09-25:
I see, but as stated the function doesn't get called, even if you uncomment. In simple words the keyboard is working again without a change to P4wnP1's boot process.
So whatever magic you have done, it must have been something else.

So in case there's a keyboard issue, there have to be some steps to reproduce.

gonebad1222, 2017-09-25:
fresh install.... only changed that....
it could be a keyboard issue

mame82, 2017-09-25:
As the code changes highlight above don't have a real effect, I assume something went wrong with driver installation on your first attempt on Windows end.

I was able to produce such a behavior on Windows 7 one time, this was when Windows update jumped in on automatic driver installation. This lead to a situation where it takes minutes to bring up the keyboard driver.

So could you please add in some details of the devices detected by device manager with unmodified code?

gonebad1222, 2017-09-25:
the problem is I let it sit for a 12 hour period nothing happened... I checked my P's and Q's

gonebad1222, 2017-09-25:
still unable to get the backdoor up..

mame82, 2017-09-25:
Copied, could you retry and give details on the detected devices. As you're working with Kali, too, it would be nice to receive some dmesg output after connecting P4wnP1 to Kali (the part when it comes to USB detection)

gonebad1222, 2017-09-25:
the problem, also is kali doesn't detect my machines usb ports on the sides. more and more i'm thinkning its this fujitsu corprate handmedown laptop i got for a steal.. but it doesn't work on other machines i've tried aswell. give me a few i've allotted time today to figure this out will get back with you.

gonebad1222, 2017-09-26:
so fresh install... nothing was changed.. just chose the keyboard payload... nothing... again... tried on several machines.... could it be a faulty pi???? i mean i am no novice computer programmer. just new to pi and i cant wait to see pythons full potential but i am a quick learner. Its late and i will give you the dmesg reports tomorrow.

mame82, 2017-10-04:
Anything new on this?

closed 2017-10-10 by mame82