RoganDawes/P4wnP1 #119: I have created an Auto Installer for P4wnP1 (Under Heavy Development) But it works.

gonebad1222, 2017-09-26:
I would like some of the mods here to take a look at a file it is a .sh file that will install P4wnP1 from beginning to end even set up the network interface automatically ssh in and then run the necessary commands for installation. since i can't seem to get my pi working correctly i figured there were other ways i could help out let me know!

jcstill, 2017-09-26:
Interesting idea. May be useful to others in your situation, where they can't get the P4wnP1 working properly.
Although, I am not sure that this would be put into the project. Reason being that mame82 has already stated that there wont be a pre-built image [(wiki link)](https://github.com/mame82/P4wnP1-Wiki/wiki/FAQ#the-setup-isnt-easy-will-there-be-a-prebuilt-image) and [(FAQ.md link)](https://github.com/mame82/P4wnP1/blob/master/FAQ.md#the-setup-isnt-easy-will-there-be-a-prebuilt-image). I feel that this and a pre-built image are almost the same thing.
BTW where can I see this file?

gonebad1222, 2017-09-26:
right here
[pwnpisetup.zip](https://github.com/mame82/P4wnP1/files/1331912/pwnpisetup.zip)

gonebad1222, 2017-09-26:
Also I will be adding interaction features soon to take arguments so you can copy and paste the IP and interaction features so you can  choose the usb0 ,eth0 as how you are connected with the PI and then choose how you want to connect the pi through your computers Internet. it "gets mad" bc it doesn't like "dumb ssh" basically a auto ssh program but it only throws warnings and it also removes the IP if you are doing another install on a new or existing PI.

mame82, 2017-10-12:
Should be moved to WiKi, but seems to specific and a bit buggy

closed 2017-10-12 by mame82