RoganDawes/P4wnP1 #120: Turkish Keyboard Bug

MeantCoder, 2017-09-26:
Hi,

I'm tested keyboard input and count some wrong keys

in payload  ">" character out "Ç" (UPPER CASE) 

How to fix this ?

**Just tested turkish keyboard layout

Swiftb0y, 2017-09-26:
Do you mein like you wanzend to Type > and ended up with Ç, or the other way around?

MeantCoder, 2017-09-26:
@Swiftb0y No, Just typed Ç Like this ; http://prntscr.com/gq27kt

Swiftb0y, 2017-09-26:
This is something for mame to fix, but according to the [mapping](https://github.com/mame82/duckencoder.py/blob/master/resources/tr.properties), the ascii 3E (>) will be pressed by shift and period. I don't know if that helps you, and have you made sure to set the lang to tr in the **payload** you are using?

MeantCoder, 2017-09-26:
Yes lang tr in payload other characters typing good i think this big source duckyencoder

MeantCoder, 2017-09-26:
*Not big "bug" sorry for auto-complate

Swiftb0y, 2017-09-27:
Might be, but try to mess with the mentioned mapping and see if the current key combination might be the issue.

mame82, 2017-10-04:
The key sequence responible for printing out `>` on TR layout is defined here:

https://github.com/mame82/duckencoder.py/blob/9dbae3ddb70b8eab76258312bfe4462942b4bfc8/resources/tr.properties#L68

This means when an US 105 key keyboard is attached, the PERIOD key and SHIFT have to be pressed to produce output of an `>` while the target keyboard layout is set to TR.

Obviously this is a wrong key combination. Now there are two ways to find out about the correct combination, maybe you could help solving this:

1) Attach an US keyboard and set the OS keyboard layout to TR. Some key combination should produce a `>`. The keys from the US keyboard which produce the correct results are the ones which have to be put into `tr.resource` at the line highlighted above.

2) A physical TR keyboard could be attached and the OS keboard layout set to US. Now pressing the key sequence which normaly outputs the `>` should produce another character (hopefully an ASCII char).
Now `us.resource` could be searched for this character `ASCII_XX` which corresponds to an USB key sequence in this file. This again is the correct sequence which should be used to replace SHIFT+PERIOD in `tr.resource`

mame82, 2017-10-04:
I hope the steps aren't to confusing, I can try myself but this needs some time

henshin, 2017-10-05:
Hi @mame82 !
I was trying with the portuguese keyboard but I'm also having a missing key. In my case it's the dash "-".  The original payload looks like this:
```
$h=(Get-Process -Id $pid).MainWindowHandle;$ios=[Runtime.InteropServices.HandleRef];$hw=New-Object $ios (1,$h);$i=New-Object $ios (2,0);(([reflection.assembly]::LoadWithPartialName("WindowsBase")).GetType("MS.Win32.UnsafeNativeMethods"))::SetWindowPos($hw,$i,{0},{1},{2},{3},{4})'.format(x, y, cx, cy, flags)
```
The FireStage1 prints this:
```
$h=(GetProcess Id $pid).MainWindowHandle;$ios=[Runtime.InteropServices.HandleRef];$hw=NewObject $ios (1,$h);$i=NewObject $ios (2,0);(([reflection.assembly]::LoadWithPartialName("WindowsBase")).GetType("MS.Win32.UnsafeNativeMethods"))::SetWindowPos($hw,$i,100,100,80,80,16512)
```
Which then obviously causes Powershell to throw an error regarding the GetProcess method being invalid.
I looked in the keyboard mapping on [pt.properties](https://github.com/mame82/duckencoder.py/blob/9dbae3ddb70b8eab76258312bfe4462942b4bfc8/resources/pt.properties) file but it seems correctly assigned to the US slash key.
```
ASCII_2D = KEY_SLASH,
ASCII_5F = KEY_SLASH, MODIFIERKEY_SHIFT	
```
Any idea what could be the problem?

mame82, 2017-10-12:
`pt` is fixed for turkish keyboard I need some community support or it has to be delayed

netm4li, 2018-02-20:
Hello,

My lab setup has Windows 7 Turkish with a physical Turkish keyboard plugged in.

When I try to use **hid_backdoor_remote** payload, final CMD text comes as:

> Object *Fşıex )no IOçStreamReader)no IOçCompressıonçGZıpStream))no IOçMemoryStream *A ')öğConvertüŞŞFromBase64Strıng)+b===öğIOçCompressıonçCompressıonModeüŞŞDecompress===çReadToEnd)=şexıt

The only topic I could found on web for Turkish language for P4wnP1 is this thread. So:

1. When I dig into **hid_backdoor_remote.txt** I found that this line:
[https://github.com/mame82/P4wnP1/blob/master/payloads/hid_backdoor_remote.txt#L135](url)

2. And it refers to following line which shows that **tr** (Turkish) keyboard is not supported:
[https://github.com/mame82/P4wnP1/blob/master/hidtools/backdoor/config.txt#L7](url)

@mame82 as per your last comment, I am willing to give support if you lead me.