RoganDawes/P4wnP1 #121: Win10_LockPicker Hash will onyl collected with single username

pixelwasser, 2017-09-26:
Hello,

I have found maybe the next issue. Currently it seems that the Win10_LockPicker payload is only able to collect hashes from single user Name like "DemoUser" if you there is a space in the Username like "Demo User" the script tries to collect the hash from "Demo" which doesn't exist on the machine.

Sincerley
Jan

pixelwasser, 2017-09-28:
Target was a Windows 10 1607

Swiftb0y, 2017-09-28:
Even though you explained your issue very clearly, I fear mame is the only one that can help you. So be patient for a view days and you'll probably get a good answer.

mame82, 2017-10-11:
The root cause which allowed this payliad to work was fixed with KB4041691. So I'll give no further support on this, but a writeup is on the way

closed 2017-10-11 by mame82