RoganDawes/P4wnP1 #122: windows 10 lock screen hash problem

gamahawk97, 2017-09-26:
hi 
i installed P4wnp1 on raspberry pi zero w.... when i boot pi i see error. 
 what is the problems ? where am i wrong?
problems in picture;

[image]

thanks for helping now :)

mame82, 2017-09-26:
What's the output of `uname -r` ?

gamahawk97, 2017-09-26:
the output is 4.9.51+

mame82, 2017-09-26:
Nice, please connect P4wnP1 to internet again and run
```
sudo apt-get update && sudo apt-get -y upgrade
```

This should hopefully upgrade livssl to 1.1 which is missing on your installation

gamahawk97, 2017-09-26:
ok i  will trying but i dont connect  internet ip addr is always seen 172.24.0.1 how can i fix this?
and thanks a lot for helping :)

mame82, 2017-09-26:
Use the WiFi_client options from the network_only payload, to connect to a give WiFi or follow this tutorial:

https://youtu.be/QEWaIoal5qU

gamahawk97, 2017-09-27:
i solved all problems(started from the beginning) it works perfect :D . and your tutorial is very  helpful thanks a lot again ...

Swiftb0y, 2017-09-27:
you should close it then and ill add an entry to the wiki

closed 2017-10-04 by mame82