RoganDawes/P4wnP1 #124: Question: How can I contribute?

DeadPackets, 2017-09-28:
Is writing payloads enough? Or can I help with something else? I don't know python but I'd love to support the project in any way possible. Any suggestions?

(Thinking of adding some NodeJS onto the Pi Zero for a WebSocket connection back to a server somewhere on the internet)

Swiftb0y, 2017-09-28:
It would be great to have a new contributor. Please look into the [wiki](https://github.com/mame82/P4wnP1-Wiki/wiki). It's still WIP but it might answer some of your questions.
The ability to reach back to a server got already achieved with a reverse autoSSH tunnel, so mame probably wouldn't merge that. Since P4wnP1 is not very powerful in terms of computing, we highly suggest you don't use complex tools that take long to startup, use a lot of resources and take up large amounts of storage. Also, please don't use to many dependcies that might slow p4wnp1 down. 
While we would like to have features like Enterprise WPA support, these features require modifing the core which is still under heavy development and any contributions to that in the near future will only slow down development due to headaches from merge conflicts and the other problems that come with it.

DeadPackets, 2017-09-28:
> Addition: we are currently moving the wiki to readthedocs. It would be
> great to have a native speaker look over the wiki and PR the changes once
> they got corrected.

I would definitely love to go help produce the docs and even proofread them
as well!

Swiftb0y, 2017-09-28:
That's great. I will tell you once we are done moving.

mame82, 2017-10-04:
I'll close this, as everything is answered by SwiftB0y and an according section was added to the WiKi

closed 2017-10-04 by mame82