RoganDawes/P4wnP1 #125: set UK Keyboard

sq34k, 2017-09-28:
Hi Im from the uk and have to set the keyboard for the whole install to uk somehow lol,
I'm fairly new to your software and have noticed uk is not in the supported languages, I noticed this when it started putting "#" instead of "\" I was just wondering how to go about adding it or unlocking it, any help would be much appreciated,
also after this i can help to start writing Payloads, for you too, I want to contribute,

2.0 another query is how to go about adding the hardware switch for the pi zero w.
I presume it is something to do with the stem being able to go from power and load rpi, to straight usb, but unsure of its look and application...

Swiftb0y, 2017-09-28:
I'm not quite sure what your issue is.
Do you mean the keyboard is typing # for no reason or " instead of #?

The problems with the switch is, that while it might be useful for some in certain conditions, most people won't buy a switch and solder it to their Pi. There are two two good implementations:
* A button is used to switch to payloads and the state gets indicated on a bunch of leds (small but flashy)
* A DIP switch is used to enable different payloads (ugly, big)
In both implementations, either a reboot or editing of the core are required which are both suboptimal since most people don't want to reboot the Pi and modifing the core will only slow down development due to merge conflicts. 

I'm glad you are interested in this project and it would great to have a native speaker for the wiki on board.

sq34k, 2017-09-28:
Hi thanks for your reply
My issue seems to be the pi and district boot in defauly US and for some of
the payloads it seems to stick in US even after I difine GB in the settings
I was basically wondering if there was a way to set the default keyboard
form
O - US to
18 - GB

As to the switch I will look into it as it seems like more of a cool
feature than an important one I guess aesthetics mean more to me hahaha

And a native speaker erm is that for proof reading?. Sorry I know I have a
lot of questions but I want to have a good foundation of this project so I
can help more I'm currently writing a payload to strip CHROME_PASS and copy
to device there are others I've made too which are just fun lol

Swiftb0y, 2017-09-28:
This is a reoccurring issue. The `lang` gets overridden in most payloads back to us. Just look at your currently selected payload in the payload/ subdirectory and change that. 
And the more questions the better. But I wrote a [troubleshooting page](https://GitHub.com/mame82/P4wnP1-Wiki/wiki/Troubleshooting) for issues like that. 
I believe we already have a password stealer that's the "hackin9" payload.
But I believe even fun troll payloads could have a place in P4wnP1.

sq34k, 2017-09-28:
Well my joke one played Mortys evil theme in the background from rick and
Morty it was a marker for an early payload to I knew it had activated lol

And ok I will look through those issues trouble shoots too

Me and a friend are working on a troll one that starts the
matrixscreensaver then turns off keyboard and mouse lol

Thank-you for talking to me.

Swiftb0y, 2017-09-28:
Those are probably doable via duckyscript. So you can PR the script itself if you want.

mame82, 2017-09-28:
> I was basically wondering if there was a way to set the default keyboard form O - US to 18 - GB 

It seems you're referring to the HID backdoor. The server has a payload independent language setting.

The config file is here:

https://github.com/mame82/P4wnP1/blob/master/hidtools/backdoor/config.txt

Change the `keyboard_lang` option to gb

sq34k, 2017-09-29:
thanks this has helped alot and i was looking into the switch and
applications and i cam accross this

https://thepihut.com/collections/raspberry-pi-store/products/wittypi-mini-rtc-power-management-for-raspberry-pi

in theory you could write scripts set off hours exen days after initiall
injection, and also when you reboot it could swap from setup to FireStage1
too i was quite interesting, and compared to the bash bunny i see this
having real life application for security testing and also admin work,
another example is people who send their computers in because they are
locked out and need local it help this platform has so much potential on
both sides of the HAT lol

but yeah if i got one of these do you think it could run as a said "arming
switch" sorry again
and the native speaker thing if thats proof reading i would be happy to help

=^.^=

Swiftb0y, 2017-09-30:
This HAT would be really useful for power management. But the switches are supposed to be able to select payloads and so on. If there is a switch HAT. I might consider buying one and add support in P4wnP1 for it.

closed 2017-10-04 by mame82