RoganDawes/P4wnP1 #127: duckhid types ">>" before writing a string

Hugand, 2017-10-01:
I've been using the P4wnP1 and recently I noticed that if I just want to write something, let's say "Hello World!", on the notepad for example, it writes ">>Hello World!" instead of just "Hello World!". Is it normal and what can I do to fix this??

Hugand, 2017-10-01:
And also it ignores the "-" character for example in the portuguese layout

mame82, 2017-10-04:
Could you provide steps to reproduce?
Are you using `SendKeys` from the backdoor payload or `outhid` from an SSH shell or payload?

mame82, 2017-10-04:
Sorry, I ignored the `duckhid` in the title. Could you please provide a sample of a DuckyScript which causes this error?

Hugand, 2017-10-05:
# assign custom drivelabel to UMS
UMSLABEL="HUG4ND"
fatlabel $wdir/USB_STORAGE/image.bin $UMSLABEL

# =============================
# USB setup
# =============================
USB_VID="0x1d6b"    # Vendor ID
USB_PID="0x1002"    # Product ID
USE_ECM=false       # we need no Linux/Mac networking
USE_RNDIS=true      # RNDIS network device to enable hash stealing
USE_HID=true        # HID keyboard to allow entering cracked password
USE_UMS=true # enable USB Mass Storage

# use LED based HID keyboard test, to fire callback to onKeyboardUp()
HID_KEYBOARD_TEST=true

# overwrite defaul keyboard language
lang="pt"

function onKeyboardUp()
{

	cat <<- EOF | duckhid
		DELAY 2000
		STRING Hello World! -
		ENTER
		DELAY 1000
	EOF

	echo "Just testing - _ %" | outhid
		
}

Hugand, 2017-10-05:
for example

mame82, 2017-10-06:
I'll try to reproduce this next week

Hugand, 2017-10-06:
ok

Hugand, 2017-10-12:
So @mame82 , how's this issue going? Any progress?

mame82, 2017-10-12:
I'm able to reproduce and preparing fix

mame82, 2017-10-12:
Should be fixed, in fact this was caused by a very small typo ... only one "comma" has to be removed.

Hugand, 2017-10-13:
Oh nice! I'll try it when I get home.

Hugand, 2017-10-14:
It works like a charm!!! Great job!

mame82, 2017-10-14:
Thx for reporting

closed 2017-10-14 by mame82