RoganDawes/P4wnP1 #130: Disconnecting from the internet - Route Default ( RaspiBerry PI W )

stephanodesouza, 2017-10-04:
How is it possible at the time of loading the payload to download a file that is in RaspiBerry PI W? When I connect it on the target the internet of the target stops working, because it creates a default route for its IP. This way I can not download from an external server.

mame82, 2017-10-04:
I'm not sure if I interpreted your question correctly.
It seems you're refering to creation of a route which propagates the IP address of P4wnP1 as default gateway. This behavior is driven by the `ROUTE_SPOOF` option which applies for Ethernet over USB only (RNDIS/CDC ECM) and is disabled except for the LockPicker payload.

mame82, 2017-10-12:
Closed because of missing feedback

closed 2017-10-12 by mame82

JoshGreen1337, 2017-10-15:
Hi I'm having a similar issue, when I plug a freshly installed P4wnP1 into my ubuntu 17.04 a new route is created with the P1 as the default route on my laptop.

Looking further in "init_usb_ethernet.sh" the option "ROUTE_SPOOF" is being correctly set as false and the generated dnsmasq config is simply:
```
bind-interfaces
port=0
interface=usb1
listen-address=172.16.0.1
dhcp-range=172.16.0.2,172.16.0.2,255.255.255.252,5m

dhcp-leasefile=/tmp/dnsmasq.leases
dhcp-authoritative
log-dhcp
```
Even manually setting the real gateway as a lower metric or removing the P1's route from my table doesn't work; the P1's default route gets readded. It seems to be a NetworkManager behaviour, especially since setting the route to "Link-Local Only" fixes the gateway issue.

I'm not sure how to fix this whilst keeping the dhcp server on the P1 active. Currently plugging the P1 in kills the internet connection on the main device. Any suggestions?

mame82, 2017-10-15:
I have to investigate this ... seems dnsmasq sets default gw + DNS even if the option isn't provided.

Acvording to this
https://superuser.com/questions/306121/i-dont-want-my-dhcp-to-be-a-default-gateway
... the respective options need to be set as empty, but I need to test this