RoganDawes/P4wnP1 #132: Questions about backdoor_hid

camilitox, 2017-10-05:
Since in Backdoor_hid, "Stage1.ps1" is a Powershell script, how can we add other powershell script? (for example the powersploit suite). and is it possible to launch a powershell shell instead of a cmd shell?
Thank you for your excellent work, sorry for my bad English
Merge state

Swiftb0y, 2017-10-06:
You can Embed you powershell Script in Dome Duckyscript and pipe it to duckhid from the ssh session. Alternatively you can Also use that duckyscript from the hid_backdoor cli with `SendDuckyScript` when you place the script in the DuckyScript/ folder.

camilitox, 2017-10-06:
And how would be the right way to do it?: I tried and it did not work "STRING powershell -NoP -NonI -W Hidden -Exec Bypass .\script.ps1" and "STRING powershell -NoP -NonI -W Hidden -Exec Bypass /home/pi/P4wnP1/DuckyScripts/shell.ps1"

mame82, 2017-10-12:
> STRING powershell -NoP -NonI -W Hidden -Exec Bypass /home/pi/P4wnP1/DuckyScripts/shell.ps1

Typing this out to a target couldn't work, as it knows nothing about P4wnP1's internal file structure.
I suggest doing some research on HID keyboard attacks, follow the advice of @Swiftb0y  and check the hakin9 payload (example on how to type out a PowerShell script which is stored on P4wnP1)

closed 2017-10-12 by mame82