RoganDawes/P4wnP1 #133: SSH no longer works.

mattstorm360, 2017-10-07:
After i install P4wnP1 on a pi Zero i'm able to SSH into it through the USB using putty 172.16.0.1:22 but for some reason after i leave the pi unplugged and alone for a while when i plug the pi back in i am unable to SSH into it and putting SSH into the boot folder dose not work. After reformatting the sd card and reinstalling everything i am able to SSH through a USB again but will lose the ability to SSH into it if i don't use it soon enough. I'm not sure if this is a problem with P4wnP1 or with RASPBIAN STRETCH LITE.

mame82, 2017-10-12:
Seems you have configured you USB interface with a static IP, but the configuration gets overwritten (maybe a DHCP client is running on the interface)

Codez33, 2017-10-22:
I am having the same problem. Any fix?

mame82, 2018-04-07:
Please try the ready-to-go image provided here https://github.com/mame82/P4wnP1/releases

closed 2018-04-07 by mame82

wlan6mon, 2018-04-13:
mame82 , Hey I just Downloaded your image and it works very well on my pi , so i have one question to ask can i change acsess point name from here?

WIFI_ACCESSPOINT=true
WIFI_ACCESSPOINT_NAME="P4wnP1"
WIFI_ACCESSPOINT_AUTH=true # Use WPA2_PSK if true, no authentication if false
WIFI_ACCESSPOINT_CHANNEL=6
WIFI_ACCESSPOINT_PSK="MaMe82-P4wnP1"
WIFI_ACCESSPOINT_IP="172.24.0.1" # IP used by P4wnP1
WIFI_ACCESSPOINT_NETMASK="255.255.255.0"
WIFI_ACCESSPOINT_DHCP_RANGE="172.24.0.2,172.24.0.100" # DHCP Server IP Range

mame82, 2018-04-13:
Exactly, but you have to do it in the actual payload in use. If the payload **doesn't define** these options, the ones from `setup.cfg` are used

wlan6mon, 2018-04-13:
Theres Another Question I have Two PI's So I Can't ssh In Second PI with pi@172.16.0.1 . What Can i do with it?

Swiftb0y, 2018-04-13:
If you use both Pi's on the same machine and connect them both via the same route your PC won't be able to reach the other one. Possible solutions are: 
*  set USB_PID, USB_PID and IF_IP(+IF_DHCP_RANGE) to something different on one of them. 
*  connect to one Pi via USB and to the other one via WIFI_AP.
*  connect them both to a wifi accesspoint with your machine. 

I hope I understood your question as intended.

wlan6mon, 2018-04-14:
How Can I fix this problem?
[image]

Swiftb0y, 2018-04-14:
Im Notfall that familiar with the internals of nano but are you editing the .swp on purpose?

mame82, 2018-04-14:
Can you paste the outpu of `df -h`

wlan6mon, 2018-04-14:
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       3.4G  1.8G  1.4G  57% /
devtmpfs        213M     0  213M   0% /dev
tmpfs           218M     0  218M   0% /dev/shm
tmpfs           218M  3.1M  215M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           218M     0  218M   0% /sys/fs/cgroup
/dev/mmcblk0p1   42M   21M   21M  51% /boot
tmpfs            44M     0   44M   0% /run/user/1000

mame82, 2018-04-14:
So I suggest deleting the `setup.cfg.swp` file.
Please avoid opening issues, which refer to generic Linux tools or raspbian (not related to P4wnP1).
Lock file issues with nano aren't related to P4wnP1 and I'm pretty sure they can be solved with a bit of searching.