RoganDawes/P4wnP1 #134: Pi Zero W - Raspbian stretch - WiFi AP issues - not reading setup.cfg correctly?

r3d-dy3-n0-2, 2017-10-09:
After running the install script, I booted up with an OTG cable to enable interactive mode.  I was able to connect the the WiFi access point created without issues.

However when I booted without the OTG cable I was not able to access the WiFi access point, it was showing up correctly but the password was not being accepted.

I connected using the USB network interface and started hostapd in debug mode to check what was happening when a attempt to connect to the access point was made.  The connection is rejected with the following error:

`nl80211: set_key failed; err=-22 Invalid argument`

I then attempted to change the access point settings in setup.cfg but these did not stick on reboot.  If I change the payload settings these are picked up alright.

In the end I sourced the setup.cfg file at the beginning of $wdir/boot/boot_p4wnP1

`source $wdir/setup.cfg`

Now the WiFi access point is accessible when running with OTG and without.  I have not dug in further to see why the payloads are being set correctly and WiFi was not.  This "hack" has allowed me to test the tool further which is great,  but not sure it is the correct solution?

r3d-dy3-n0-2, 2017-10-09:
Okay on further investigation it appears the issues was that most of the shell scripts in $wdir/boot/ were not given executable permissions.  Adding executable permissions to the scripts resolved the problems.

I was able to delete the line below that I added to boot_p4wnP1 and everything is working fine.

`source $wdir/setup.cfg`

Can these scripts be made executable by the install script, or have the executable bit set on them when checked in.

mame82, 2017-10-10:
> I then attempted to change the access point settings in setup.cfg but these did not stick on reboot. If I change the payload settings these are picked up alright.

Intended behavior. `setup.cfg` is meant to provide default setting, unless they're overwritten by more specific payload settings. Remove the WIFI options from the payload, if the ones from `setup.cfg` should be used.

> In the end I sourced the setup.cfg file at the beginning of $wdir/boot/boot_p4wnP1

This is already done here:
https://github.com/mame82/P4wnP1/blob/master/boot/boot_P4wnP1#L43
... referencing here:
https://github.com/mame82/P4wnP1/blob/master/boot/init_config.sh

> Okay on further investigation it appears the issues was that most of the shell scripts in $wdir/boot/ were not given executable permissions. Adding executable permissions to the scripts resolved the problems.

This shouldn't be needed. None of the scripts is executed directly, but they're all sourced in to an executable boot script. Making them executable could lead to errors, because most of them rely on variables from the "master script", which would be missing if they are executed "stand-alone".

It seems the root cause of your problem was altering WiFi settings in `setup.cfg` instead of the payload.

closed 2017-10-12 by mame82