RoganDawes/P4wnP1 #135: "GetClientProc" coammnd not woking

careyer, 2017-10-09:
Hi there,

first of all: "Terrific work!" This rules so much!
Everything working fine. Hoever I found the "GetClientProc" command not working. It throws an error that some function is not found (sorry i don't have the correct error message at hand right now).

Keep up the excellent work!
Cheers!
careyer

mame82, 2017-10-10:
Sorry for this, I've dropped the client side implementation of `getClientProc` for rework, so the server and complains. For now use something like `shell` and issue `tasklist`.

I paused working on the backdoor to analyze other techniques (nexmon drivers + Bluetooth LE).

Yhanks for the compliments

careyer, 2017-10-10:
Thanks! Yeah this is a suitable workaround! Just thought I'd let you know. I am curious about what's next to come! Thanks for the terrific work!

closed 2017-10-12 by mame82