RoganDawes/P4wnP1 #138: Feature request: keylogger

careyer, 2017-10-11:
Hi there,

i am currently in the works to hide the P4wnP1 loaded Pi0W together with a small USB hub in a regular keyboard. (once I succeed I'll post some pictures here).

This makes for a very stealthy unit which can easily be deployed and stay unnoticed for quite some time. BTW: I am working as a penetration tester and as a security consultant I am also into awareness trainings - i.e I am not interested in doing evil with this device. I just want to have it for showcase purposes on my presentations and trainings to show-off some state of the art tools (not the good old rubber ducky again). 

Besides all of the terrific (and terrifying) features P4wnP1 offers as of now the last bit missing to make it even more perfect is a keylogger feature. This way it would yield for a perfect data ex filtration and attack platform. Since the P4wnP1 is hidden inside a regular keyboard anyway there must be a possibility to grab the user input and safe it to the MicroSD card inside the Pi and forward all keystrokes via USB to the PC.

However this seems to be a bit complicated (at least too complicated for me) to be solved easily. The only USB-Port on the Pi0W is already occupied. It would somehow need to have a USB for input (where the keyboard is connected) and one for output (this could be the same output as used nowadays).

@mame82 : Is there any chance to get this feature added in the future? Any idea how one could achieve this?

Thx in advance!
Keep up the excellent work - U ROCK!

Swiftb0y, 2017-10-11:
The Problem with These approaches is that the Pi (slave) and the master would have to be connected directly (to my knowledge), which also why any other raspberry model with a integrated USB-Hub won't work with P4wnP1.
I think if you really want to try this, get a pHAT that works as an USB Master, while not interfering with the built in USB-Controller. You can then use a keylogger with the ability to register callbacks to construct some duckyscript and pipe it to the onboard duckencoder, or to assemble the raw messages directly (look into https://GitHub.com/MaMe82/duckencoder/)

Swiftb0y, 2017-10-11:
P.S. Since this is a very specific use case and requires soldering and extra hardware, don't expect Mame to use his limited time on this. We would be happy if this was a payload contribution by you. 
Visit the [wiki](https://p4wnp1.readthedocs.io/) for more information on contributing.

mame82, 2017-10-11:
I won't implement a (hardware based) keylogger, due to the restrictions highlighted by @Swiftb0y.
A microcontroller based approach (USB host controller + keyboard device emulation = keyboard MitM) would be the best solution for this, essentially this ends up with a setup already available on market.

Anyway, let me point you in a direction which could help to achieve your goal:
The hakin9_tutorial payload demoes how to write data generated by a PowerShell script to built-in USB storage (which ends up in an image on the SD). As you might already know, there PowerShell post exploitation frameworks out there, including PS based keyloggers. So you could depliy the using the techniques show cased with the mentioned payload, to store results to USB.

Of  course this approach is limited to Windows and without further modifications you only grab keystrokes from the user session  (no LockScreen input etc. ), but it should be enough for the purpose of presenting it to users in awareness trainings.

For engagements this basic solution could be refined (add in UAC bypass,  use a separate extraction channel for keylogger data or hide it on fileservice for example).

It should be mentioned that some research has been published on side channel attacks to USB hub (sniffing devices on the same hub using cross talking on both: data lines and power lines), which could nicely fit into your setup. This of course couldn't be done with P4wnP1 (low level hardware attack, based on voltage meassuring)

mame82, 2017-10-11:
P.S. You should maybe have a look into IronHID. The author deployed a similar attack, but implanted a microcontroller based solution into a real keyboard. It could be used to inject keystrokes via WiFi, too. As far as I remember there was a use case where he fetched the real output from the hardware via wired connection directly from the keyboard matrix.

In my opinion it isn't the best idea to bring in a new keyboard device during a pentest. A commercial hardware keylogger is far better for this purpose and could be deployed on different clients, based on the needs. P4wnP1 aim isn't to replace sich tools, but to bring up new ways to get a foothold on the engagement target.

careyer, 2017-10-11:
@mame82 & @Swiftb0y : Thank you very much for your comments! I understand the limitations of the Pi0W and yes there are most likely better ways to achieve what I was searching for. I guess I will go for the PS based keylogger approach since I want to keep the build clean and easy and avoid inducing another microcontroller. Thank you for pointing me into the right direction! THX!

closed 2017-10-12 by mame82