RoganDawes/P4wnP1 #140: Recent changes to install.sh causing problems

5yn-1, 2017-10-12:
I encountered some issues when re-installing P4wnP1 today. The installation would complete without any errors; however when I rebooted the pi, I was met with an endless stream of:
./home/pi/P4wnP1/boot/init_usb_ethernet.sh  line 58 /sys/class/net/usb0/carrier no such file
./home/pi/P4wnP1/boot/init_usb_ethernet.sh  line 78 /sys/class/net/usb1/carrier no such file
Most system functions were limited; not allowing me to connect via ssh, or use Tab to autocomplete.
There was also mention of the file system being read-only whenever I tried to make changes.

I found that simply removing the new lines from install.sh before initially running it allowed the pi to boot properly, and function fully.
I spent a few hours attempting other solutions, but nothing else seemed to have any effect.

I hope that this is able to help anyone encountering the same issue.

mame82, 2017-10-12:
Have to verify that, sorry for the inconvenience

5yn-1, 2017-10-12:
No worries.
Let me know if you would like me to provide any addition information.

jryd2000, 2017-10-12:
It looks like you just need to install gawk during the install.sh script.  The new lines for fsck are referencing a function in awk that doesn't exist on the base install of raspbian stretch lite.

mame82, 2017-10-12:
Should be solved with last commit

5yn-1, 2017-10-12:
Thanks! I'll give that a try when I'm back at my desk and let you know the results.

5yn-1, 2017-10-13:
It installed with no errors, and all features seem to be fully functioning!
Thanks for the quick fix!

closed 2017-10-13 by mame82

mroxso, 2018-04-06:
I am facing this issue again with latest raspbian and latest P4wnP1.

Pooky-s, 2018-04-15:
Having the same error :c

alienelephant, 2018-05-18:
I had the same error using https://github.com/mame82/P4wnP1/commit/62b5c40d9461e4b80998c54830625d6c855833f0. 

Confirmed installing gawk prior to running the install script fixed the issue, no more boot loop.

mame82, 2018-05-18:
I recommend using the pre-built image from releases ... no need to run the install script with this

KnightSlayerr, 2018-10-25:
Problem is, the pre-bult Pi image of Kali with P4wnP1 installed has the same error - /sys/class/net/usb0/carrier: no such file or directory