RoganDawes/P4wnP1 #145: SSH Problems

DieBambusleitung, 2017-10-16:
Hey,

I have a fresh installed Raspbian Stretch Lite (Version 2017-09-07) and installed P4wnP1, but I'm not able to ssh into the pi. I already tried an older raspbian version, but it also didn't work. I also tried a different USB port, so it can't be that... The LED on the BOOTED pi off is this normal? I need some help there.

mame82, 2017-10-16:
There was an error in the DHCP config which should be fixed by the latest commit (some minutes ago)

mame82, 2017-10-17:
Does your problem remain with latest commit?

mame82, 2017-10-17:
According the LED: The `network_only` payload doesn't run a command to enable LED. So having the LED off is fine

DieBambusleitung, 2017-10-17:
Thanks! Now it works!