RoganDawes/P4wnP1 #146: Can't use USB Mass Storage

Hugand, 2017-10-16:
I've already fresh installed P4wnP1 3 times and when I run a payload that enables it, like the hakin9_tutorial one, but the drive simply doesn't show.

mame82, 2017-10-17:
Did you get any details from device manager

Hugand, 2017-10-17:
I didnt see that. When I get home I can give more information about it.

Hugand, 2017-10-17:
Everything looks normal, the computer recognizes the pi but doesn't let me save things in it. Here is a screenshot of how it shows on my PC.
[image]

Swiftb0y, 2017-10-17:
Stupid question:
Have you tried another cable?

Hugand, 2017-10-17:
yes

Swiftb0y, 2017-10-17:
Could you post a screenshot from the device manager and disk utility?

Hugand, 2017-10-17:
ok

Hugand, 2017-10-17:
[image]
[image]

Swiftb0y, 2017-10-22:
Sorry man, no Idea. Could you post the output from `sudo journalctl-u P4wnP1.service`?

Hugand, 2017-10-22:
-- Logs begin at Thu 2016-11-03 17:16:43 UTC, end at Mon 2017-10-16 22:38:23 UTC. --
Oct 16 22:37:14 MAME82-P4WNP1 systemd[1]: Starting P4wnP1 Startup Service...
Oct 16 22:37:14 MAME82-P4WNP1 bash[171]: =================================== P4wnP1 startup ===========================================
Oct 16 22:37:14 MAME82-P4WNP1 bash[171]: P4wnP1: Init LED control...
Oct 16 22:37:14 MAME82-P4WNP1 bash[171]: P4wnP1: Loading config ...
Oct 16 22:37:15 MAME82-P4WNP1 bash[171]: open: No such file or directory
Oct 16 22:37:16 MAME82-P4WNP1 bash[171]: P4wnP1: Initializing USB gadget ...
Oct 16 22:37:16 MAME82-P4WNP1 bash[171]: /home/pi/P4wnP1/boot/init_usb.sh: line 142: echo: write error: No such file or directory
Oct 16 22:37:17 MAME82-P4WNP1 bash[171]: crw------- 1 root root 242, 0 Oct 16 22:37 /dev/hidg0
Oct 16 22:37:17 MAME82-P4WNP1 bash[171]: crw------- 1 root root 242, 1 Oct 16 22:37 /dev/hidg1
Oct 16 22:37:17 MAME82-P4WNP1 bash[171]: crw------- 1 root root 242, 0 Oct 16 22:37 /dev/hidg0
Oct 16 22:37:17 MAME82-P4WNP1 bash[171]: crw------- 1 root root 242, 1 Oct 16 22:37 /dev/hidg1
Oct 16 22:37:17 MAME82-P4WNP1 bash[171]: P4wnP1: Checking for WiFi capabilities ...
Oct 16 22:37:18 MAME82-P4WNP1 bash[171]: P4wnP1: Seems WiFi module is present !
Oct 16 22:37:19 MAME82-P4WNP1 dnsmasq[259]: started, version 2.76 DNS disabled
Oct 16 22:37:19 MAME82-P4WNP1 dnsmasq[259]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify
Oct 16 22:37:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: DHCP, IP range 172.24.0.2 -- 172.24.0.100, lease time 5m
Oct 16 22:37:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: DHCP, sockets bound exclusively to interface wlan0
Oct 16 22:37:19 MAME82-P4WNP1 bash[171]: USB OTG off, going on with P4wnP1 boot
Oct 16 22:37:19 MAME82-P4WNP1 bash[171]: P4wnP1: ... USB gadget initialized
Oct 16 22:37:19 MAME82-P4WNP1 bash[171]: P4wnP1: Initializing Ethernet over USB...
Oct 16 22:37:19 MAME82-P4WNP1 bash[171]: CDC ECM or RNDIS active. Check which interface has to be used via Link detection
Oct 16 22:37:19 MAME82-P4WNP1 bash[171]: .Link detected on usb0
Oct 16 22:37:19 MAME82-P4WNP1 systemd[1]: Started P4wnP1 Startup Service.
Oct 16 22:37:19 MAME82-P4WNP1 bash[171]: Waiting for HID keyboard to be usable...
Oct 16 22:37:21 MAME82-P4WNP1 bash[171]: 0
Oct 16 22:37:22 MAME82-P4WNP1 bash[171]: P4wnP1: Creating DHCP configuration for Ethernet over USB...
Oct 16 22:37:23 MAME82-P4WNP1 dnsmasq[368]: started, version 2.76 DNS disabled
Oct 16 22:37:23 MAME82-P4WNP1 dnsmasq[368]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify
Oct 16 22:37:23 MAME82-P4WNP1 dnsmasq-dhcp[368]: DHCP, IP range 172.16.0.2 -- 172.16.0.2, lease time 5m
Oct 16 22:37:23 MAME82-P4WNP1 dnsmasq-dhcp[368]: DHCP, sockets bound exclusively to interface usb0

Hugand, 2017-10-22:
for what I've seen I dont have the USB_STORAGE/image.bin on the pi as I saw from "Oct 16 22:37:16 MAME82-P4WNP1 bash[171]: /home/pi/P4wnP1/boot/init_usb.sh: line 142: echo: write error: No such file or directory"

Swiftb0y, 2017-10-22:
And there you have your Solution: run the lines in the p4wnp1 installer again that go start with something like `dd if=/dev/zero ...` in the P4wnP1 same location as the installer and see if that fixes it.

Hugand, 2017-10-23:
Ok, thanks!

Hugand, 2017-10-23:
It works! Thanks for the help!

closed 2017-10-23 by mame82