RoganDawes/P4wnP1 #147: P4wnP1 incorporated into a Keyboard

careyer, 2017-10-16:
Not exactly an issue/bug report but just some experience from incorporating a P4wnP1 into a USB keyboard.

I used a cheap USB 2.0 High Speed Hub from ebay (#301820219992) in order to be able to connect both keyboard and Pi0W internally to the USB cable heading towards the PC. 
Everything turned out nicely but the solution has some flaws that I wanna share: 

1.  on first conenct a short notice that a new USB-Hub 2.0 is installed is given
2.  even though it should be a USB 2.0 Highpseed Hub it seems to be of some slow sort. Windows pops up a notice that the connected device could be faster on a Highspeed port (or is the Pi0W acting as a USB 3.0 device?)
[image]
3.  the keystroke injection is considerably slower than usual
4.  since the Pi0W is hidden beneath the handrest the user will notice that the handrest gets considerably warmer after some time

Keyboard opened up:
[image]
A bit of USB-Hub stripping, soldering and hot-glue fixing
[image]
Put back together:
[image]
Looks like brand new in its original cardbox box
[image]

mame82, 2017-10-17:
Very nice. P4wnP1 is running on USB2.0, btw.

careyer, 2017-10-17:
Thanks for your compliment! It turned out quite well - the only dissatisfiying thing is this USB 2.0 notice. I have tested again on a USB 2.0 only host-port on my PC as well and the notification is still displayed. Seems like the HUB is only running at USB 1.0 speed for some unknown reason (it is a USB 2.0 devices though).  That would als explain why the keyboard injection works WAY(!) slower right now. I will debug further and share my findings here if anyone is interested

careyer, 2017-10-19:
**UPDATE:** Finally figured out what the problem was with the USB-Hub was. After hours and hours of debugging it turned out that the item was a fake USB 2.0 hub. Only the String "USB-Hub 2.0" is used as the USB device name whereas the chipset used is only a cheap USB 1.1 chipset . Damn - there is so much faked stuff arround on ebay lately. (Just recently bought some faked Samsung MicroSD Cards, now the same thing with commodity products like USB hubs) GRRRR!!!

I changed the USB Hub for a real USB 2.0 version now and everything works brilliantly! I also got rid of the problem that the handrest would heat up after a while by replacing the Pi0W to another location inside of the keyboard. Everything stays nice and unnoticeably cool now! ;-)
[image]
[image]

Cheers!
Keep up the excellent work!

onehitwonderos, 2018-02-04:
Great work! Can you tell me what USB hub you used? It looks really small

careyer, 2018-02-05:
@onehitwonderos : It is just the PCB from a random "wire harness" style USB-Hub. Pretty much something like this... https://www.amazon.de/DIGITUS-High-Speed-4-Ports-Stecker-Netzteil/dp/B004BU8HWI
This style of USB hub can be easily disassembled and has a rather small footprint.

ch3lmi, 2018-07-07:
Excellent idea @careyer ! 
I finally took the time to assemble mine and also put the pi zero and the up in the top of the keyboard, most likely at the same place as you did. Just had to rip off some inner plastic from the keyboard to fit the electronics in.

Thanks for the inspiration

careyer, 2019-03-12:
Today I had the chance the play a bit with neat digital X-Ray technology. 

Enjoy! : 
[image]

Swiftb0y, 2019-03-12:
Good luck getting that past the TSA ;)