RoganDawes/P4wnP1 #148: Laptop doenst recongize Raspi Zero W as keybord

bufo24, 2017-10-17:
Hello,
I did connect my raspi zero w to my laptop using the usb connector, but my laptop doesn't find it.

Can anyone help me out?

Swiftb0y, 2017-10-18:
There are many possible reasons why it may not work. We would need the following things:
On Windows:
* Screenshots of the disk utility 
* Screenshots of the device-Manager
* Screenshot of the available (W)lan networks
If possible in English or German.
On Linux/OSX:
* `lspci`
* `ifconfig`
If you have only one OS, the results from the other OS aren't important.

mame82, 2017-10-18:
Which USB connector are you using + has your USB cable data lines connected?

bufo24, 2017-10-18:
I'm using a micro-usb to usb cable, it goes from my laptop to the data port of my raspi zero w

Swiftb0y, 2017-10-18:
Have you tried SSHing into your pi via wifi?

bufo24, 2017-10-18:
Yes, and it works

Swiftb0y, 2017-10-18:
Have you tried to use `echo Hello World! | outhid` to pipe raw ascii to the keyboard?

Swiftb0y, 2017-10-18:
Or how did you test it?

bufo24, 2017-10-18:
I can login to ssh, but if I type echo Hello World! | outhid , there happens nothing...

Swiftb0y, 2017-10-18:
Ok. Have you checked that you enabled HID in the setup.cfg as well as in the payload you are using?
And could you post the Pi's output of `journalctl -u P4wnP1.service`?

bufo24, 2017-10-18:
My output is:

-- Logs begin at Thu 2016-11-03 17:16:44 UTC, end at Tue 2017-10-17 21:37:02 UTC. --
Oct 17 21:17:05 MAME82-P4WNP1 systemd[1]: Starting P4wnP1 Startup Service...
Oct 17 21:17:06 MAME82-P4WNP1 bash[170]: =================================== P4wnP1 startup ===========================================
Oct 17 21:17:06 MAME82-P4WNP1 bash[170]: P4wnP1: Init LED control...
Oct 17 21:17:08 MAME82-P4WNP1 bash[170]: P4wnP1: Loading config ...
Oct 17 21:17:09 MAME82-P4WNP1 bash[170]: P4wnP1: Initializing USB gadget ...
Oct 17 21:17:10 MAME82-P4WNP1 bash[170]: crw-rw-rw- 1 root root 242, 0 Oct 17 21:17 /dev/hidg0
Oct 17 21:17:10 MAME82-P4WNP1 bash[170]: crw-rw-rw- 1 root root 242, 1 Oct 17 21:17 /dev/hidg1
Oct 17 21:17:10 MAME82-P4WNP1 bash[170]: crw-rw-rw- 1 root root 242, 0 Oct 17 21:17 /dev/hidg0
Oct 17 21:17:10 MAME82-P4WNP1 bash[170]: crw-rw-rw- 1 root root 242, 1 Oct 17 21:17 /dev/hidg1
Oct 17 21:17:10 MAME82-P4WNP1 bash[170]: P4wnP1: Checking for WiFi capabilities ...
Oct 17 21:17:11 MAME82-P4WNP1 bash[170]: P4wnP1: Seems WiFi module is present !
Oct 17 21:17:11 MAME82-P4WNP1 dnsmasq[282]: started, version 2.76 DNS disabled
Oct 17 21:17:11 MAME82-P4WNP1 dnsmasq[282]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify
Oct 17 21:17:11 MAME82-P4WNP1 dnsmasq-dhcp[282]: DHCP, IP range 172.24.0.2 -- 172.24.0.100, lease time 5m
Oct 17 21:17:11 MAME82-P4WNP1 dnsmasq-dhcp[282]: DHCP, sockets bound exclusively to interface wlan0
Oct 17 21:17:11 MAME82-P4WNP1 bash[170]: USB OTG off, going on with P4wnP1 boot
Oct 17 21:17:11 MAME82-P4WNP1 bash[170]: P4wnP1: ... USB gadget initialized
Oct 17 21:17:11 MAME82-P4WNP1 bash[170]: P4wnP1: Initializing Ethernet over USB...
Oct 17 21:17:12 MAME82-P4WNP1 bash[170]: CDC ECM or RNDIS active. Check which interface has to be used via Link detection
Oct 17 21:17:12 MAME82-P4WNP1 systemd[1]: Started P4wnP1 Startup Service.
Oct 17 21:17:12 MAME82-P4WNP1 bash[170]: ....Waiting for HID keyboard to be usable...
Oct 17 21:17:20 MAME82-P4WNP1 hostapd[277]: wlan0: STA d0:df:9a:42:12:20 IEEE 802.11: associated
Oct 17 21:17:20 MAME82-P4WNP1 hostapd[277]: wlan0: STA d0:df:9a:42:12:20 IEEE 802.11: disassociated
Oct 17 21:17:25 MAME82-P4WNP1 hostapd[277]: wlan0: STA d0:df:9a:42:12:20 IEEE 802.11: associated
Oct 17 21:17:25 MAME82-P4WNP1 hostapd[277]: wlan0: STA d0:df:9a:42:12:20 IEEE 802.11: disassociated
Oct 17 21:17:31 MAME82-P4WNP1 hostapd[277]: wlan0: STA d0:df:9a:42:12:20 IEEE 802.11: associated
Oct 17 21:17:31 MAME82-P4WNP1 hostapd[277]: wlan0: STA d0:df:9a:42:12:20 RADIUS: starting accounting session 59E67357-00000002
Oct 17 21:17:31 MAME82-P4WNP1 hostapd[277]: wlan0: STA d0:df:9a:42:12:20 WPA: pairwise key handshake completed (RSN)
Oct 17 21:17:31 MAME82-P4WNP1 bash[170]: ...............................................................................................................................
Oct 17 21:17:31 MAME82-P4WNP1 dnsmasq-dhcp[282]: 2517470061 available DHCP range: 172.24.0.2 -- 172.24.0.100
Oct 17 21:17:31 MAME82-P4WNP1 dnsmasq-dhcp[282]: 2517470061 vendor class: MSFT 5.0
Oct 17 21:17:31 MAME82-P4WNP1 dnsmasq-dhcp[282]: 2517470061 client provides name: DESKTOP-O219GB0
Oct 17 21:17:34 MAME82-P4WNP1 dnsmasq-dhcp[282]: 2517470061 DHCPDISCOVER(wlan0) d0:df:9a:42:12:20
Oct 17 21:17:34 MAME82-P4WNP1 dnsmasq-dhcp[282]: 2517470061 tags: wlan0
Oct 17 21:17:34 MAME82-P4WNP1 dnsmasq-dhcp[282]: 2517470061 DHCPOFFER(wlan0) 172.24.0.2 d0:df:9a:42:12:20
Oct 17 21:17:34 MAME82-P4WNP1 dnsmasq-dhcp[282]: 2517470061 requested options: 1:netmask, 3:router, 6:dns-server, 15:domain-name,
Oct 17 21:17:34 MAME82-P4WNP1 dnsmasq-dhcp[282]: 2517470061 requested options: 31:router-discovery, 33:static-route, 43:vendor-encap,
Oct 17 21:17:34 MAME82-P4WNP1 dnsmasq-dhcp[282]: 2517470061 requested options: 44:netbios-ns, 46:netbios-nodetype, 47:netbios-scope,
Oct 17 21:17:34 MAME82-P4WNP1 dnsmasq-dhcp[282]: 2517470061 requested options: 121:classless-static-route, 249, 252
Oct 17 21:17:34 MAME82-P4WNP1 dnsmasq-dhcp[282]: 2517470061 next server: 172.24.0.1
Oct 17 21:17:34 MAME82-P4WNP1 dnsmasq-dhcp[282]: 2517470061 sent size:  1 option: 53 message-type  2
Oct 17 21:17:34 MAME82-P4WNP1 dnsmasq-dhcp[282]: 2517470061 sent size:  4 option: 54 server-identifier  172.24.0.1

And which HID do I have to enable?

Swiftb0y, 2017-10-18:
Seems like a driver issue to me because of `Oct 17 21:17:12 MAME82-P4WNP1 bash[170]: ....Waiting for HID keyboard to be usable...`
Is the last relevant log. Try changing the VID and PID in the payload. And please post the current content of your setup.cfg

bufo24, 2017-10-18:
This is my setup.cnf:

And is this driver issue only on my laptop or at everyones?

#!/bin/sh

###########################
# General config
#       these are the default settings
#       the setting are only used, if not defined in the payload itself
###########################

# USB setup
# ---------------------------
# Make sure to change USB_PID if you enable different USB functionality in order
# to force Windows to enumerate the device again
USB_VID="0x1d6b"        # Vendor ID
USB_PID="0x0137"        # Product ID

USE_ECM=true            # if true CDC ECM will be enabled
USE_RNDIS=true          # if true RNDIS will be enabled
USE_HID=false            # if true HID (keyboard) will be enabled
USE_HID_MOUSE=true            # if true HID mouse will be enabled
USE_RAWHID=false         # if true a raw HID device will be enabled
USE_UMS=false           # if true USB Mass Storage will be enabled

# ===========================================
# Network and DHCP options USB over Ethernet
# ===========================================

# We choose an IP with a very small subnet (see comments in README.rst)
IF_IP="172.16.0.1" # IP used by P4wnP1
IF_MASK="255.255.255.252"
IF_DHCP_RANGE="172.16.0.2,172.16.0.2" # DHCP Server IP Range

ROUTE_SPOOF=false # set two static routes on target to cover whole IPv4 range
WPAD_ENTRY=false # provide a WPAD entry via DHCP pointing to responder

# ============================
# WiFi options (only Pi Zero W)
# ============================
WIFI_REG=US # WiFi regulatory domain (if not set accordingly, WiFi channels are missing)

# Access Point Settings
# ---------------------

WIFI_ACCESSPOINT=true
WIFI_ACCESSPOINT_NAME="P4wnP1"
WIFI_ACCESSPOINT_PSK="MaMe82-P4wnP1"
WIFI_ACCESSPOINT_IP="172.24.0.1" # IP used by P4wnP1
WIFI_ACCESSPOINT_NETMASK="255.255.255.0"
WIFI_ACCESSPOINT_DHCP_RANGE="172.24.0.2,172.24.0.100" # DHCP Server IP Range
WIFI_ACCESSPOINT_HIDE_SSID=false # use to hide SSID of WLAN (you have to manually connect to the name given by WIFI_ACCESSPOINT_NAME)

# WiFi Client Settings
# --------------------

WIFI_CLIENT=false       # enables connecting to existing WiFi (currently only WPA2 PSK)
                        # example payload: wifi_connect.txt
                        # Warning: could slow down boot, because:
                        #       - scan for target network is issued upfront
                        #       - DHCP client is started and waits for a lease on WiFi adapter
                        # Note: if WIFI_ACCESSPOINT is enabled, too:
                        #       - P4wnP1 tries to connect to the given WiFi
                        #       - if connection fails, the AccessPoint is started instead
WIFI_CLIENT_SSID="Accespoint Name" # name of target network
WIFI_CLIENT_PSK="AccessPoint password" # passphrase for target network
WIFI_CLIENT_STORE_NETWORK=false # unused right now, should be used to store known networks, but priority has to be given if multiple known networks are present
WIFI_CLIENT_OVERWRITE_PSK=true # unused right now, in case the network WIFI_CLIENT_STORE_NETWORK is set an existing PSK gets overwritten

[... 34 lines cut ...]
BLUETOOTH_NAP=false # Enable Bluetooth NAP to SSH in via Bluetooth
BLUETOOTH_NAP_PIN=1337 # unused, PIN authentication currently not working (custom agent for bluez 5 needed)
BLUETOOTH_NAP_IP="172.26.0.1" # IP used by P4wnP1
BLUETOOTH_NAP_NETMASK="255.255.255.0"
BLUETOOTH_NAP_DHCP_RANGE="172.26.0.2,172.26.0.100" # DHCP Server IP Range

# =============================
# Settings for USB Mass Storage
# =============================

# =====================
# Payload selection
# =====================

#PAYLOAD=network_only.txt
#PAYLOAD=hid_mouse.txt # HID mouse demo: Shows different ways of positioning the mouse pointer, using P4wnP1's MouseScript languag
#PAYLOAD=hid_backdoor_remote.txt # AutoSSH "reachback" version of hid_backdoor (see payload comments for details)
#PAYLOAD=wifi_connect.txt
#PAYLOAD=stickykey/trigger.txt # Backdoor Windows LockScreen with SYSTEM shell, triggered by NUMLOCK, trigger SCROLLLOCK to revert the changes
#PAYLOAD=hakin9_tutorial/payload.txt # steals stored plain credentials of Internet Explorer or Edge and saves them to USB flash drive (for hakin9 tutorial)
#PAYLOAD=Win10_LockPicker.txt # Steals NetNTLMv2 hash from locked Window machine, attempts to crack the hash and enters the plain password to unlock the machin on success
#PAYLOAD=hid_backdoor.txt # under (heavy) development
#PAYLOAD=hid_frontdoor.txt # HID covert channel demo: Triggers P4wnP1 covert channel console by pressing NUMLOCK 5 times on target (Windows)
#PAYLOAD=hid_keyboard.txt # HID keyboard demo: Waits till target installed keyboard driver and writes "Keyboard is running" to notepad
#PAYLOAD=hid_keyboard2.txt # HID keyboard demo: triggered by CAPS-, NUM- or SCROLL-LOCK interaction on target

Swiftb0y, 2017-10-18:
It seems that you don't have any payload selected with USE_HID (Keyboard) disabled, but I'm not sure if that information conflicts with the one from the log.

bufo24, 2017-10-18:
Also if I turn that on, I've got the same problem...

Swiftb0y, 2017-10-19:
You have to change the VID/PID so Windows "rescans" the device.

UnknownPlayer99, 2017-10-28:
Hi, I have the same problem and i tried changing the VID/PID to something else but it still does not work. Do I have to change it to something specific or will anything work?

Swiftb0y, 2017-10-28:
Normally anything should work. Have you checked that USE_HID is enabled in  the **payload** you are using?
If yes, please post the log (`sudo journalctl -u P4wnP1.service`)

UnknownPlayer99, 2017-10-28:
Oct 27 00:21:50 MAME82-P4WNP1 bash[166]: =================================== P4wnP1 startup ===========================================
Oct 27 00:21:50 MAME82-P4WNP1 bash[166]: P4wnP1: Init LED control...
Oct 27 00:21:51 MAME82-P4WNP1 bash[166]: P4wnP1: Loading config ...
Oct 27 00:21:52 MAME82-P4WNP1 bash[166]: P4wnP1: Initializing USB gadget ...
Oct 27 00:21:54 MAME82-P4WNP1 bash[166]: crw-rw-rw- 1 root root 243, 0 Oct 27 00:21 /dev/hidg0
Oct 27 00:21:54 MAME82-P4WNP1 bash[166]: crw-rw-rw- 1 root root 243, 1 Oct 27 00:21 /dev/hidg1
Oct 27 00:21:54 MAME82-P4WNP1 bash[166]: crw-rw-rw- 1 root root 243, 2 Oct 27 00:21 /dev/hidg2
Oct 27 00:21:54 MAME82-P4WNP1 bash[166]: crw-rw-rw- 1 root root 243, 0 Oct 27 00:21 /dev/hidg0
Oct 27 00:21:54 MAME82-P4WNP1 bash[166]: crw-rw-rw- 1 root root 243, 1 Oct 27 00:21 /dev/hidg1
Oct 27 00:21:54 MAME82-P4WNP1 bash[166]: crw-rw-rw- 1 root root 243, 2 Oct 27 00:21 /dev/hidg2
Oct 27 00:21:54 MAME82-P4WNP1 bash[166]: P4wnP1: Checking for WiFi capabilities ...
Oct 27 00:21:54 MAME82-P4WNP1 bash[166]: P4wnP1: Seems WiFi module is present !
Oct 27 00:21:55 MAME82-P4WNP1 dnsmasq[259]: started, version 2.76 DNS disabled
Oct 27 00:21:55 MAME82-P4WNP1 dnsmasq[259]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify
Oct 27 00:21:55 MAME82-P4WNP1 dnsmasq-dhcp[259]: DHCP, IP range 172.24.0.2 -- 172.24.0.100, lease time 5m
Oct 27 00:21:55 MAME82-P4WNP1 dnsmasq-dhcp[259]: DHCP, sockets bound exclusively to interface wlan0
Oct 27 00:21:55 MAME82-P4WNP1 bash[166]: USB OTG off, going on with P4wnP1 boot
Oct 27 00:21:55 MAME82-P4WNP1 bash[166]: P4wnP1: ... USB gadget initialized
Oct 27 00:21:55 MAME82-P4WNP1 bash[166]: P4wnP1: Initializing Ethernet over USB...
Oct 27 00:21:55 MAME82-P4WNP1 systemd[1]: Started P4wnP1 Startup Service.
Oct 27 00:21:55 MAME82-P4WNP1 bash[166]: Waiting for HID keyboard to be usable...
Oct 27 00:22:13 MAME82-P4WNP1 hostapd[254]: wlan0: STA e0:b9:a5:f5:44:85 IEEE 802.11: associated
Oct 27 00:22:13 MAME82-P4WNP1 hostapd[254]: wlan0: STA e0:b9:a5:f5:44:85 IEEE 802.11: disassociated
Oct 27 00:22:16 MAME82-P4WNP1 hostapd[254]: wlan0: STA e0:b9:a5:f5:44:85 IEEE 802.11: associated
Oct 27 00:22:16 MAME82-P4WNP1 hostapd[254]: wlan0: STA e0:b9:a5:f5:44:85 IEEE 802.11: disassociated
Oct 27 00:22:19 MAME82-P4WNP1 hostapd[254]: wlan0: STA e0:b9:a5:f5:44:85 IEEE 802.11: associated
Oct 27 00:22:19 MAME82-P4WNP1 hostapd[254]: wlan0: STA e0:b9:a5:f5:44:85 RADIUS: starting accounting session 59F27C23-00000002
Oct 27 00:22:19 MAME82-P4WNP1 hostapd[254]: wlan0: STA e0:b9:a5:f5:44:85 WPA: pairwise key handshake completed (RSN)
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 available DHCP range: 172.24.0.2 -- 172.24.0.100
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 vendor class: MSFT 5.0
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 client provides name: Filip-Laptop
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 DHCPREQUEST(wlan0) 172.24.0.56 e0:b9:a5:f5:44:85
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 tags: wlan0
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 DHCPACK(wlan0) 172.24.0.56 e0:b9:a5:f5:44:85 Filip-Laptop
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 requested options: 1:netmask, 3:router, 6:dns-server, 15:domain-name,
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 requested options: 31:router-discovery, 33:static-route, 43:vendor-encap,
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 requested options: 44:netbios-ns, 46:netbios-nodetype, 47:netbios-scope,
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 requested options: 121:classless-static-route, 249, 252
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 next server: 172.24.0.1
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 sent size:  1 option: 53 message-type  5
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 sent size:  4 option: 54 server-identifier  172.24.0.1
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 sent size:  4 option: 51 lease-time  5m
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 sent size:  4 option: 58 T1  2m30s
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 sent size:  4 option: 59 T2  4m22s
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 sent size:  4 option:  1 netmask  255.255.255.0
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 sent size:  4 option: 28 broadcast  172.24.0.255
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 sent size:  4 option:  3 router  172.24.0.1
Oct 27 00:22:19 MAME82-P4WNP1 dnsmasq-dhcp[259]: 792491087 sent size: 15 option: 81 FQDN  03:ff:ff:46:69:6c:69:70:2d:4c:61:70:74:6f...

mame82, 2017-10-28:
Could you please provide the content of `/tmp/device_hid_keyboard`.

UnknownPlayer99, 2017-10-28:
/dev/hidg0

mame82, 2017-10-28:
Okay, open editor on windows host and issue `echo test | outhid` on P4wnP1 console to check if the HID gadget is working at all, please

UnknownPlayer99, 2017-10-28:
Nothing happens when i execute the command.

tomadeb, 2017-11-15:
I have the same problem here. 
- brand new raspbian 
- git clone ...
- sudo bash install.sh
- reboot 
pi@MAME82-P4WNP1:~/P4wnP1 $ cat /tmp/device_hid_keyboard
/dev/hidg0
pi@MAME82-P4WNP1:~/P4wnP1 $ 

"echo test | outhid" just hangs

Swiftb0y, 2017-11-15:
Do your logs differ from the one above?

tomadeb, 2017-11-17:
nope. Although I only want keyboard emulation.

-- Logs begin at Thu 2016-11-03 17:16:43 UTC, end at Fri 2017-11-17 13:55:58 UTC. --                                                                                                                                             
Nov 15 14:17:06 MAME82-P4WNP1 systemd[1]: Starting P4wnP1 Startup Service...                                                                                                                                                     
Nov 15 14:17:07 MAME82-P4WNP1 bash[175]: =================================== P4wnP1 startup ===========================================                                                                                          
Nov 15 14:17:07 MAME82-P4WNP1 bash[175]: P4wnP1: Init LED control...                                                                                                                                                             
Nov 15 14:17:07 MAME82-P4WNP1 bash[175]: P4wnP1: Loading config ...                                                                                                                                                              
Nov 15 14:21:23 MAME82-P4WNP1 bash[175]: P4wnP1: Initializing USB gadget ...                                                                                                                                                     
Nov 15 14:21:24 MAME82-P4WNP1 bash[175]: crw------- 1 root root 242, 0 Nov 15 14:21 /dev/hidg0                                                                                                                                   
Nov 15 14:21:24 MAME82-P4WNP1 bash[175]: crw------- 1 root root 242, 0 Nov 15 14:21 /dev/hidg0                                                                                                                                   
Nov 15 14:21:24 MAME82-P4WNP1 bash[175]: P4wnP1: Checking for WiFi capabilities ...                                                                                                                                              
Nov 15 14:21:24 MAME82-P4WNP1 bash[175]: P4wnP1: Seems WiFi module is present !

tomadeb, 2017-11-17:
I think this is possibly a kernel/driver problem as I have tried to create the gadget manually and have had the same issue ( hanging echo to /dev/hidg0). However, it seems I am not familiar enough with this to get any further.

Swiftb0y, 2017-11-17:
Run install.sh again and make sure the kernel patch gets installed successfully. It's near the end of the installation.

tomadeb, 2017-11-17:
pi@MAME82-P4WNP1:~/P4wnP1 $ sudo bash install.sh
Testing Internet connection and name resolution...
...[pass] Internet connection works
Testing if the system runs Raspbian Jessie or Stretch...
...[pass] Pi seems to be running Raspbian Jessie or Stretch
Backing up resolv.conf
Installing needed packages...
Hit:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
autossh is already the newest version (1.4e-3).
bluez-tools is already the newest version (0.2.0~20140808-5).
bridge-utils is already the newest version (1.5-13+deb9u1).
ethtool is already the newest version (1:4.8-1).
git is already the newest version (1:2.11.0-3+deb9u2).
hostapd is already the newest version (2:2.4-1+deb9u1).
inotify-tools is already the newest version (3.14-2).
python-dev is already the newest version (2.7.13-2).
screen is already the newest version (4.5.0-6).
sqlite3 is already the newest version (3.16.2-5).
bluez is already the newest version (5.43-2+rpt2+deb9u2).
dnsmasq is already the newest version (2.76-5+rpt1+deb9u1).
python-pip is already the newest version (9.0.1-2+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Installing needed python additions...
Requirement already satisfied: pycrypto in /usr/lib/python2.7/dist-packages
Requirement already satisfied: pydispatcher in /usr/local/lib/python2.7/dist-packages
Disabeling unneeded services to shorten boot time ...
update-rc.d: error: cannot find a LSB script for ntp
Create udev rule for HID devices...
Enable SSH server...
Checking network setup..
Entry for manual configuration of RNDIS interface found
Entry for manual configuration of CDC ECM interface found
Unpacking John the Ripper Jumbo edition...
Branch jtr_stretch set up to track remote branch jtr_stretch from origin.
Switched to a new branch 'jtr_stretch'
Configure Responder...
Creating 128 MB image for USB Mass Storage emulation
128+0 records in
128+0 records out
134217728 bytes (134 MB, 128 MiB) copied, 13.9306 s, 9.6 MB/s
mkfs.fat 4.1 (2017-01-24)
Disable FSCK on boot ...
Enable autologin for user pi...
Enable overlay filesystem for USB gadgedt suport...
Enable kernel module for USB Composite Device emulation...
Removing all former modules enabled in /boot/cmdline.txt...
Installing kernel update ...
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date
Generating keypair for use with AutoSSH...
/home/pi/P4wnP1/ssh/keys/P4wnP1_id already exists.
Overwrite (y/n)? y
... keys created

Use "/home/pi/P4wnP1/ssh/pushkey.sh"
in order to promote the public key to a remote SSH server

====================================================================================
If you came till here without errors, you shoud be good to go with your P4wnP1...
...if not - sorry, you're on your own, as this is work in progress

Attach P4wnP1 to a host and you should be able to SSH in with pi@172.16.0.1 (via RNDIS/CDC ECM)

If you use a USB OTG adapter to attach a keyboard, P4wnP1 boots into interactive mode

If you're using a Pi Zero W, a WiFi AP should be opened. You could use the AP to setup P4wnP1, too.
          WiFi name:    P4wnP1
          Key:          MaMe82-P4wnP1
          SSH access:    pi@172.24.0.1 (password: raspberry)

  or via Bluetooth NAP:    pi@172.26.0.1 (password: raspberry)

Go to your installation directory. From there you can alter the settings in the file 'setup.cfg',
like payload and language selection

If you're using a Pi Zero W, give the HID backdoor a try ;-)

You need to reboot the Pi now!
====================================================================================

tomadeb, 2017-11-17:
Same result

tomadeb, 2017-11-17:
I also tested with the latest kernel 4.9.62+ and got the same result, ie outhid just hangs indefinitely.

tomadeb, 2017-11-17:
```
pi@MAME82-P4WNP1:~ $ sudo dmesg 
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.51+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1036 Fri Sep 22 19:43:40 BST 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 114688
[    0.000000] free_area_init_node: node 0, pgdat c09185e4, node_mem_map db010000
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:A1:BC:25 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=9b9c2452-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 435924K/458752K available (5953K kernel code, 489K rwdata, 1952K rodata, 404K init, 734K bss, 14636K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
                   lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0xc0008000 - 0xc05d8748   (5954 kB)
                     .init : 0xc0843000 - 0xc08a8000   ( 404 kB)
                     .data : 0xc08a8000 - 0xc0922728   ( 490 kB)
                      .bss : 0xc0922728 - 0xc09d9f48   ( 735 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000031] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000075] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000177] bcm2835: system timer (irq = 27)
[    0.000652] Console: colour dummy device 80x30
[    0.001431] console [tty1] enabled
[    0.001488] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060356] pid_max: default: 32768 minimum: 301
[    0.060833] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060903] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062203] Disabling memory control group subsystem
[    0.062412] CPU: Testing write buffer coherency: ok
[    0.062514] ftrace: allocating 21724 entries in 64 pages
[    0.179205] Setting up static identity map for 0x8200 - 0x8238
[    0.181291] devtmpfs: initialized
[    0.190714] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.191230] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.191330] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.192653] pinctrl core: initialized pinctrl subsystem
[    0.194173] NET: Registered protocol family 16
[    0.196657] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.206378] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.206471] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.206611] Serial: AMBA PL011 UART driver
[    0.209566] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.210366] uart-pl011 20201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.258823] bcm2835-dma 20007000.dma: DMA legacy API manager at dc80d000, dmachans=0x1
[    0.261703] SCSI subsystem initialized
[    0.262032] usbcore: registered new interface driver usbfs
[    0.262215] usbcore: registered new interface driver hub
[    0.262455] usbcore: registered new device driver usb
[    0.269270] raspberrypi-firmware soc:firmware: Attached to firmware from 2017-09-22 19:50
[    0.271237] clocksource: Switched to clocksource timer
[    0.325374] VFS: Disk quotas dquot_6.6.0
[    0.325573] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.325936] FS-Cache: Loaded
[    0.326329] CacheFiles: Loaded
[... 121 lines cut ...]
[   11.444200] usb0: MAC 42:63:65:66:43:21
[   11.445926] usb1: HOST MAC 42:63:65:12:34:56
[   11.446032] usb1: MAC 42:63:65:65:43:21
[   11.446092] dwc2 20980000.usb: bound driver configfs-gadget
[   13.497126] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   13.497150] brcmfmac: power management disabled
[   16.421430] IPv6: ADDRCONF(NETDEV_UP): usb0: link is not ready
[   16.471719] IPv6: ADDRCONF(NETDEV_UP): usb1: link is not ready
[   16.587676] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[... cut ...]
[   22.431775] Bluetooth: L2CAP socket layer initialized
[   22.431833] Bluetooth: SCO socket layer initialized
[   22.448857] Bluetooth: HCI UART driver ver 2.3
[   22.448878] Bluetooth: HCI UART protocol H4 registered
[   22.448885] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   22.449092] Bluetooth: HCI UART protocol Broadcom registered
[   23.031374] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   23.031386] Bluetooth: BNEP filters: protocol multicast
[   23.031419] Bluetooth: BNEP socket layer initialized
```

tomadeb, 2017-11-17:
```
pi@MAME82-P4WNP1:~ $ cat P4wnP1/setup.cfg 
#!/bin/sh

###########################
# General config
#   	these are the default settings
#	the setting are only used, if not defined in the payload itself
###########################

# USB setup
# ---------------------------
# Make sure to change USB_PID if you enable different USB functionality in order
# to force Windows to enumerate the device again
USB_VID="0x1d6b"        # Vendor ID
USB_PID="0x0137"        # Product ID

USE_ECM=true            # if true CDC ECM will be enabled
USE_RNDIS=false          # if true RNDIS will be enabled
USE_HID=true            # if true HID (keyboard) will be enabled
USE_HID_MOUSE=false            # if true HID mouse will be enabled
USE_RAWHID=false         # if true a raw HID device will be enabled
USE_UMS=false           # if true USB Mass Storage will be enabled

# ===========================================
# Network and DHCP options USB over Ethernet
# ===========================================

# We choose an IP with a very small subnet (see comments in README.rst)
IF_IP="172.16.0.1" # IP used by P4wnP1
IF_MASK="255.255.255.252"
IF_DHCP_RANGE="172.16.0.2,172.16.0.2" # DHCP Server IP Range

ROUTE_SPOOF=false # set two static routes on target to cover whole IPv4 range
WPAD_ENTRY=false # provide a WPAD entry via DHCP pointing to responder

# ============================
# WiFi options (only Pi Zero W)
# ============================
WIFI_NEXMON=false # Experimental: enables the use of Nexmon driver + firmware to add monitor mode capability

WIFI_REG=GB # WiFi regulatory domain (if not set accordingly, WiFi channels are missing)

# Access Point Settings
# ---------------------

WIFI_ACCESSPOINT=false
WIFI_ACCESSPOINT_NAME="P4wnP1"
WIFI_ACCESSPOINT_PSK="MaMe82-P4wnP1"
WIFI_ACCESSPOINT_IP="172.24.0.1" # IP used by P4wnP1
WIFI_ACCESSPOINT_NETMASK="255.255.255.0"
WIFI_ACCESSPOINT_DHCP_RANGE="172.24.0.2,172.24.0.100" # DHCP Server IP Range
WIFI_ACCESSPOINT_HIDE_SSID=false # use to hide SSID of WLAN (you have to manually connect to the name given by WIFI_ACCESSPOINT_NAME)

# WiFi Client Settings
# --------------------

WIFI_CLIENT=true 	# enables connecting to existing WiFi (currently only WPA2 PSK)
			# example payload: wifi_connect.txt
			# Warning: could slow down boot, because:
			#	- scan for target network is issued upfront
			#	- DHCP client is started and waits for a lease on WiFi adapter
			# Note: if WIFI_ACCESSPOINT is enabled, too:
			#	- P4wnP1 tries to connect to the given WiFi
			# 	- if connection fails, the AccessPoint is started instead
WIFI_CLIENT_SSID="edited" # name of target network
WIFI_CLIENT_PSK="edited" # passphrase for target network
WIFI_CLIENT_STORE_NETWORK=false # unused right now, should be used to store known networks, but priority has to be given if multiple known networks are present
WIFI_CLIENT_OVERWRITE_PSK=true # unused right now, in case the network WIFI_CLIENT_STORE_NETWORK is set an existing PSK gets overwritten

[... 35 lines cut ...]
BLUETOOTH_NAP_PIN=1337 # unused, PIN authentication currently not working (custom agent for bluez 5 needed)
BLUETOOTH_NAP_IP="172.26.0.1" # IP used by P4wnP1
BLUETOOTH_NAP_NETMASK="255.255.255.0"
BLUETOOTH_NAP_DHCP_RANGE="172.26.0.2,172.26.0.100" # DHCP Server IP Range

# =============================
# Settings for USB Mass Storage
# =============================

# =====================
# Payload selection
# =====================

#PAYLOAD=network_only.txt
#PAYLOAD=hid_mouse.txt # HID mouse demo: Shows different ways of positioning the mouse pointer, using P4wnP1's MouseScript languag
#PAYLOAD=hid_backdoor_remote.txt # AutoSSH "reachback" version of hid_backdoor (see payload comments for details)
#PAYLOAD=wifi_connect.txt
#PAYLOAD=stickykey/trigger.txt # Backdoor Windows LockScreen with SYSTEM shell, triggered by NUMLOCK, trigger SCROLLLOCK to revert the changes
#PAYLOAD=hakin9_tutorial/payload.txt # steals stored plain credentials of Internet Explorer or Edge and saves them to USB flash drive (for hakin9 tutorial)
#PAYLOAD=Win10_LockPicker.txt # Steals NetNTLMv2 hash from locked Window machine, attempts to crack the hash and enters the plain password to unlock the machin on success
#PAYLOAD=hid_backdoor.txt # under (heavy) development
#PAYLOAD=hid_frontdoor.txt # HID covert channel demo: Triggers P4wnP1 covert channel console by pressing NUMLOCK 5 times on target (Windows)
PAYLOAD=hid_keyboard.txt # HID keyboard demo: Waits till target installed keyboard driver and writes "Keyboard is running" to notepad
#PAYLOAD=hid_keyboard2.txt # HID keyboard demo: triggered by CAPS-, NUM- or SCROLL-LOCK interaction on target
```

Swiftb0y, 2017-11-18:
I can't see anything at first glance but there are some lines that seem suspicious. I'm not that familiar with the technical kernel stuff which makes gadget mode possible so you will have to wait for mame to finish his vacation until you get a proper response. 
```[    8.002924] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    8.002940] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    8.002951] usb usb1: Product: DWC OTG Controller
[    8.002961] usb usb1: Manufacturer: Linux 4.9.51+ dwc2_hsotg
[    8.002971] usb usb1: SerialNumber: 20980000.usb
[    8.004215] hub 1-0:1.0: USB hub found
[    8.004309] hub 1-0:1.0: 1 port detected
```

Seem suspicious.

Swiftb0y, 2017-11-18:
Ps can you send the settings in your payload you are using?

HectorHumanes, 2017-12-17:
I'm having the same issues commented below. Outhid just hangs indefinitely for me too.