RoganDawes/P4wnP1 #151: Powershell on target often crashes (LinkLayer timeout)

careyer, 2017-10-19:
I noticed on several target computers that the Powershell thread will often crash.
I started the FireStage1 with the "nohide" option and in all cases before the crash it displays a "LinkLayer timeout 1318" (the number may vary). 

Is there any way to avoid this? 
Thanks!

careyer, 2017-10-22:
Update: Found out that this issue is somehow related to my hardware setup 

My setup is:
Hardware Keyboard + P4wnP1 both connected via a USB2.0 Hub (and all hidden inside of the keyboard) to a Windows 7 x64 PC. (see #147).

**Szenario 1:**
When I connect my USB-Hub to the PC it will take some time until a warning balloon message appears saying that a USB-device connected could not be correctly recognized. However the message will likely disappear by itself after some seconds and the P4wnP1 Wifi-Hotspot comes up and I can successfully connect to it via SSH. However the P4wnP1 shell will not be startet (backdoor payload) and I will only see the standard bash shell.

**Szenario 2:**
I tried to connect my USB-Hub to the PC with a USB Y-Cable (Data +  Voltage only) in the following manner: First connect voltage only and let the P4wnP1 boot up.... then connect the USB-Data cable. Now no warning messages appears and when SSH-ing into to P4wnP1 I immediatly get the P4wnP1 shell (backdoor payload).

It seems like that in conjuction with the additionl USB-Hub there is some timing issue. Could it possibly help to delay the boot of the Pi for a second or so to avoid this?

Swiftb0y, 2017-10-22:
I'm not sure how you could delay the boot immediately, but if you add a `sleep <not sure if s or ms>` at the top of your setup.cfg, it will wait the specified time until setting up the usb features and wifi hotspot. I would also use this in conjunction with `led_blink` to make sure sleep is working as expected. 

Btw. Thank you very much for researching this.

Swiftb0y, 2017-10-22:
PS: it would be helpful if you'd include the services logs (`systemctl -u P4wnP1.service`) of both scenarios.

Swiftb0y, 2017-10-22:
Sorry, `journalctl -u P4wnP1.service`. Editing doesn't work for some reason.

careyer, 2017-10-22:
@Swiftb0y : Thanks... I added a "sleep 10" at the very beginning of the setup.cfg. It still gives me the "usb device could not be recognized Windows balloon message"... but... now it directly starts the P4wnP1 shell when SSH-ing in. =D 
So to say half a success ;-)

Swiftb0y, 2017-10-22:
That's good to hear. I think the logs could really explain some stuff then. It would be good if you could post them of all three of them them (scenario 1 & 2 + the sleep 10 one). And put them in backticks ;-)

careyer, 2017-10-22:
Okay, here you go - I am happy to help! =)

[10secDelay.txt](https://github.com/mame82/P4wnP1/files/1405227/10secDelay.txt)
[Szenario1.txt](https://github.com/mame82/P4wnP1/files/1405228/Szenario1.txt)
[Szenario2.txt](https://github.com/mame82/P4wnP1/files/1405229/Szenario2.txt)