RoganDawes/P4wnP1 #153: Its not auto starting and not working

Braandn, 2017-10-23:
I successfully installed P4wnP1 by following the INSTALL.md but when I plug in the Pi into my Windows 10 Computer its not working
When I connect with my phone via SSH to the P4wn Wifi the P4wn server is not started.
I tried to start it by using python /home/pi/P4wnP1/hidtools/backdoor/P4wnP1.py and the shell starts but when I type in FireStage1 its saying "Starting to type out stage1 to the target..." -> Nothing happen
The payload is set to "PAYLOAD=hit_backdoor.txt" others are marked with #
I tried to tell you my problem, please help me

HitItAll, 2017-10-23:
Is it set to hid_bwckdoor_remote.txt?

HitItAll, 2017-10-23:
It has to be the remote one, and the cable to windows must be in data slot
AND your cable must support data transfer

Braandn, 2017-10-23:
Thank you so much, it was the cable

HitItAll, 2017-10-23:
No problem!

closed 2017-10-23 by mame82