RoganDawes/P4wnP1 #155: Payload difference and Command questions..?

Braandn, 2017-10-26:
What is the difference between HID-Backdoor and HID-Backdoor-Remote and where could i change the commands for example "FireStage1" to "start" or others?

Thanks

Swiftb0y, 2017-10-26:
The remote variant uses AutoSSH to provide a reverse ssh shell which requires internet access, a properly configured server and a correct setup.cfg/hid_backdoor_remote.txt

There is no proper way to change the names of the commands but you can try editing hidtools/backdoor/P4wnP1.py from `def do_FireStage1` to `do_start`. This will break the help menu and might not work so be careful.

Swiftb0y, 2017-10-26:
PS. Don't forget the the `def` before do_start (`def do_start`)

closed 2018-04-07 by mame82