RoganDawes/P4wnP1 #157: Problem connecting to Mac or Windows emulatiion on Mac

puba, 2017-10-28:
Are configuration settings the same for windows and mac os or must they be changed in setup configuratiion? Have trouble running payloads on Mac OS and win emulation  --> I am getting only interactive shell without active payload.

Brisk4t, 2017-10-28:
Which payload are you using , try turning on USB ecm in setup.cfg

Swiftb0y, 2017-10-28:
The Setup.cfg is only there to provide defaults. The hid_backdoor.txt will override active ECM settings because Windows only has drivers for RNDIS (It's Microsofts own proprietary standard). You have to change the setting in the payload itself. But as far is I know, this won't work either since communication is handled through a covered bidirectional hid device, which probably won't go through because of the used virtualization software. If you can wait, I am working on a port of [this](https://twitter.com/mame82/status/915116312690593794).

puba, 2017-10-28:
Thanks, I really apretiate that.

closed 2017-10-28 by puba