RoganDawes/P4wnP1 #160: usb device not recognized (after many attempts)

bytedeez, 2017-11-07:
I am running raspbian Jessie kernal version 4.4.50 #970
I have tried this many times my process is as follows.
Freshly burned image on sd card plugged into spare pi zero
login as pi
setup in raspi-config
Setup wifi using sudo nano /etc/network/interfaces
apt-get update
apt-get upgrade
cd /boot/ touch ssh
apt-get install git
git clone ..P4wnP1
./install
check cmdline.txt and config.txt
modules seems to loaded via P4wnP1 script
No errors occurred during install.
shutdown -h

plug into USB gadget pi, plug in usb-pi to Win10- usb gadget not recognized.
try to ssh - fail
check device manager nothing found.
plug sd card into pi 3- can not access raspbian due to keyboard not working.
rinse, wash repeat.

Am I missing something?
I have also tried doing the same steps, while setting up usb gadget on the pi manually
I'm kinda at a lost because I had P4wnP1 working before it became so well known. Maybe back in April?

HitItAll, 2017-11-07:
Does your cable support data transfer? Also if it is too long it might not
work

Brisk4t, 2017-11-07:
Does the p4wnp1 wifi start? Also try using a different usb cable.

bytedeez, 2017-11-07:
I'm using a regular pi zero.
I've tried it using 3 different usb cables on 2 different pi0s.

mame82, 2018-04-07:
Please try the ready-to-go image provided here https://github.com/mame82/P4wnP1/releases

bytedeez, 2018-05-09:
After reburning the ready-to-go image and adding a blank file named "ssh" to the boot folder. P4wnP1 registered fine and i am not able to ssh over usb,
Thanks for your assistance!

closed 2018-05-09 by bytedeez