RoganDawes/P4wnP1 #161: No Access Point

MicroBread, 2017-11-07:
After run ./install.sh, I reboot Pi0 but can't find Wi-Fi P4wnP1. 
In win10 it's recognized as "P4wnP1 by mame82" and in OSX there is a window shows "Keyboard Setup Assistant".

I install it and reboot:
1.sudo git clone --recursive https://github.com/mame82/P4wnP1
2.cd P4wnP1
3.sudo bash install.sh
4.sudo reboot

Am I do something wrong?

HitItAll, 2017-11-07:
Do you have the RasPi0 W? You might have bought the cheaper one with no WiFi

MicroBread, 2017-11-07:
@HitItAll sorry I can't reply you by e-mail. I'm sure it's a Pi0 W

Swiftb0y, 2017-11-07:
Make sure you are using a Pi W and enable the Wifi access point in the setup.cfg as well as in the selected payload.

MicroBread, 2017-11-08:
Thanks for your reply. I didn't change setup.cfg and it seems that
ACCESSPOINT open by default. As for payload I choose
hid_backdoor_remote.txt, something wrong?

Swiftb0y, 2017-11-08:
The hid_backdoor_remote.txt is The HID_backdoor which requires Internet.
This will enable the wifi client option and autoSSH which are both not configured. 
You need to use the plain hid_backdoor.txt 

The thing with the wifi client that it should fall back into Mastermode if it doesn't find the specified Network, which it doesn't do in your case. So try switching to the normal hid_remote and post the output `sudo journalctl -u P4wnP1.service` if your issue is not resolved by switching the payload.

mame82, 2018-04-07:
Closed, no user feedback.
If issue is still present try the ready-to-go image provided here https://github.com/mame82/P4wnP1/releases

closed 2018-04-07 by mame82