RoganDawes/P4wnP1 #164: Can't install on Pi0W

vhax, 2017-11-11:
I have tried to no avail to connect to my raspberry pi zero through any means possible. I have tried the USB Ethernet gadget method, and my computer (running debian) does not detect any new interfaces. I have tried connecting it to wifi using almost every tutorial i could find on google, yet it wont connect to any wifi.
I am using a 4gb sd card, but IDK if that is an issue or not. Can someone please help?

Swiftb0y, 2017-11-11:
I think 4gb should be enough but I would still try a bigger SD card. My preferred method of setting up the Pi0W headless is modifing the config of the networkmanager and wpa_supplicant to connect to your wpa2 home network automatically. You can then enable SSH by creating an empty file on boot partition named `ssh`.

vhax, 2017-11-11:
I think it was the SD card, as i replaced it with an 8gig and it works now. But, now when I use the P4wnP1, the drivers take forever to load on windows 7

Swiftb0y, 2017-11-11:
This is most likely due an issue with your PC or the cable you are using.

vhax, 2017-11-11:
I thought it was plug and play with additional drivers needed to be installed. Also, why are all the drivers installed for something as simple for the hid_keyboard payload?

Swiftb0y, 2017-11-11:
I don't know. I just maintain the wiki. I think mame is still on vacation so you will have to wait a bit for a proper answer.

closed 2017-11-21 by mame82