RoganDawes/P4wnP1 #166: [Feature Request] Web Interface

beezTEM, 2017-11-12:
(almost) Everyone knows about SpaceHuhn's WiFi-Ducky, and his Web Interface. I believe that P4wnP1 would benefit from something similar to this. It would be helpful for those that use it in a school environment where they get chromebooks. It would also be helpful for those who have issues with command lines, and would prefer a web interface. It would be most helpful for the HID backdoor payload, as it injects keystrokes and needs interaction.

Thank You For Your Time And Consideration,
beezTEM

Swiftb0y, 2017-11-12:
The Webinterface was requested years ago and was marked as a community task. The problem is that the communication with the .net client is woven into the console front end. Splitting these into separate modules would require to much time and effort this still something for the community but not a focus of Mame. 
Just get familiar commandline
And SSH runs EVERYWHERE. There might even be some kind of portable clients you could just run of a usb drive.

lukespademan, 2017-11-15:
Some place block network traffic on port 22 and the SSH protocol. But I agree, a web interface is not a necessity.