RoganDawes/P4wnP1 #168: P4wnP1 and osX

Hyldig01, 2017-11-15:
I have been trying to get P4wnP1's hid_keyboard.txt to work with osX, without luck... every time i plug it in, the Keyboard assistance pops up saying that the keyboard cannot be identified..., Have tried to change vendor and product id to Apple keyboard id (0x05ac and 0x024f), both in setup.cfg and in hid_keyboard.txt but nu luck.... Anyone knows how to work around ?

Swiftb0y, 2017-11-15:
I have tried this to about three months ago and it worked. I've mentioned my vid and pid in some other issue but I don't have the time to search for it right know.

leperjulien, 2017-11-25:
Trying to do the same but even when i change the vid and pid to an Apple's one it doesn't work and i couldn't find the issue @Swiftb0y mentionned so if anyone knows a work around ;)

It's still a great tool for Windows anyway

Hyldig01, 2017-11-25:
I also get the keyboard assistance to pop up when I plug in my malduino or my rubber ducky, 
However both of them works “almost” perfectly and around the assistance, but for some reason the P4wnP1 won’t.  I think I’ll try to write a “delay 1500, enter, z, delay 1000, enter” in my scripts for Mac,  and hope it recognizes it and get around the! I’ll try to play with that later today. 
However my P4wnP1 is now for Windows only :-) 

You might also needs to make the “command-space” command active for your scripts to work. Found out it’s not active out of the box! And if it’s not active, you might just rename all the stuff on your desktop ;-)

leperjulien, 2017-11-25:
Same for me with a rubber ducky it pops up the Keyboard assistant but still works fine.

For the p4wnp1 , if i change the VID and PID to an Apple's one like 0x05ac and 0x021e i don't get the Keyboard Assistant anymore , but hid_keyboard.txt never get executed and nothing happend when i plug it to my Macbook.. will try again later today and see if i can get it to work.

santaklouse, 2017-11-25:
yes I using some apple keyboard ids and it preventing Keyboard assistant popups opening but I tested it only on usb rubber ducky. seems should work for pi as well

leperjulien, 2017-11-25:
Finally manage to make it work, i don't know why but hid_keyboard.txt won't work on Mac OS, but hid_backdoor.txt does work pretty well ;) I can use SendKeys and SendDuckyScript just fine and i even found online a fr.properties file for the custom layout of an Azerty keyboard so it's perfect for me ! VID : 0x05ac PID : 0x021e.

I have a question tho, if @mame82 or anybody else could help me on this one ,
When my Pi Zero W finish to boot while plugged to my Macbook it create the hotspot but when i connect to it there is the error 
```
-bash: /tmp/blink_count: Permission denied
There is no screen to be detached.
```
But i can manually type : sudo python /home/pi/P4wnp1/hidtools/backdoor/P4wnP1.py

and it will work just fine, but doest that means the Pi didn't see that the target Macbook already loaded the drivers ? That could be the same thing preventing the hid_keyboard.txt from running.. What could i do to workaround that ?

Also when plugged on windows computer , "FireStage1 1 5000" will only work 1 time over 5 , i have to enter the command many times before i got a Client connected but when i use "FireStager 1 5000 nohide" it will work everytime ! Do you know why ?

Awesome work by the way ;)

Swiftb0y, 2017-11-25:
The blink_count issue is probably due to your installation being slightly out of date. Use the Wifi_client feature to connect the Pi to a wpa2 network with Internet and use `git pull && ./install.sh` to update. This should probably also fix the issue with the P4wnP1-terminal. If not, please post the output of `sudo journalctl -u P4wnP1.service`

mame82, 2017-11-26:
Seens keyboard "driver up" detection fails on OSX.

leperjulien, 2017-11-27:
@Swiftb0y Seems weird because i already installed from git and it was only like 2 days ago..

Tried git pull && ./install.sh but it said i'm already up to date so looks like the issue is elsewhere ;)

@mame82 Yeah seems like it , i made a new file hid_backdoor_osx.txt that start P4wnP1.py using onTargetGotIP() instead because it got an IP really quick and its working perfect so far , i made a fork of your repo and added the new files (not sure if i did it well tho since i'm new to github) if anybody want to try it on his Mac.

mame82, 2017-11-27:
The proper way would be to change the detection routine. First idea was to trigger a keyboard LED (e.g. pressing num lock) till an LED change is received, which would indicate driver readiness. Problem with this approach: on early tests it produced a kernel oops when keys are send before the target driver is ready (unresponsive irq).

Thus I changed the approach to wait for a LED reset without sending a key. Windows resets the LED state as soon as the driver is ready, unfortunately OSX seems to differ in that. If this behavior isn't triggered, the onKeyboardUp function is never called, which explains the behavior. Moving the code to onTargetGotIP isn't a valid alternative for keyboard only payloads (not using network functionality)

mr0Ot, 2017-12-13:
I test it with this by the clean install:

git pull && ./install.sh

Than i start its new an i make a payload, here is the info, that you will have:

There is no screen to be detached.
pi@MAME82-P4WNP1:~ $ sudo journalctl -u P4wnP1.service
-- Logs begin at Thu 2016-11-03 17:16:43 UTC, end at Wed 2017-12-13 21:23:28 UTC. --
Dec 13 21:21:39 MAME82-P4WNP1 systemd[1]: Starting P4wnP1 Startup Service...
Dec 13 21:21:39 MAME82-P4WNP1 bash[178]: =================================== P4wnP1 startup ===========================================
Dec 13 21:21:39 MAME82-P4WNP1 bash[178]: P4wnP1: Init LED control...
Dec 13 21:21:40 MAME82-P4WNP1 bash[178]: P4wnP1: Loading config ...
Dec 13 21:21:41 MAME82-P4WNP1 bash[178]: P4wnP1: Initializing USB gadget ...
Dec 13 21:21:42 MAME82-P4WNP1 bash[178]: crw------- 1 root root 242, 0 Dec 13 21:21 /dev/hidg0
Dec 13 21:21:42 MAME82-P4WNP1 bash[178]: crw------- 1 root root 242, 1 Dec 13 21:21 /dev/hidg1
Dec 13 21:21:42 MAME82-P4WNP1 bash[178]: crw------- 1 root root 242, 2 Dec 13 21:21 /dev/hidg2
Dec 13 21:21:43 MAME82-P4WNP1 bash[178]: crw-rw-rw- 1 root root 242, 0 Dec 13 21:21 /dev/hidg0
Dec 13 21:21:43 MAME82-P4WNP1 bash[178]: crw-rw-rw- 1 root root 242, 1 Dec 13 21:21 /dev/hidg1
Dec 13 21:21:43 MAME82-P4WNP1 bash[178]: crw-rw-rw- 1 root root 242, 2 Dec 13 21:21 /dev/hidg2
Dec 13 21:21:43 MAME82-P4WNP1 bash[178]: P4wnP1: Checking for WiFi capabilities ...
Dec 13 21:21:43 MAME82-P4WNP1 bash[178]: P4wnP1: Seems WiFi module is present !
Dec 13 21:21:44 MAME82-P4WNP1 sudo[252]:     root : TTY=unknown ; PWD=/sys/kernel/config/usb_gadget/mame82gadget ; USER=root ; COMMAND=/sbin/
Dec 13 21:21:44 MAME82-P4WNP1 sudo[252]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 13 21:21:44 MAME82-P4WNP1 sudo[252]: pam_unix(sudo:session): session closed for user root
Dec 13 21:21:44 MAME82-P4WNP1 bash[178]: Try to find WiFi ESSID-of-upstream-WLAN
Dec 13 21:21:44 MAME82-P4WNP1 sudo[262]:     root : TTY=unknown ; PWD=/sys/kernel/config/usb_gadget/mame82gadget ; USER=root ; COMMAND=/sbin/
Dec 13 21:21:44 MAME82-P4WNP1 sudo[262]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 13 21:21:45 MAME82-P4WNP1 sudo[262]: pam_unix(sudo:session): session closed for user root
Dec 13 21:21:45 MAME82-P4WNP1 bash[178]: Network ESSID-of-upstream-WLAN not found
Dec 13 21:21:45 MAME82-P4WNP1 bash[178]: P4wnP1: Join present WiFi didn't succeed, failing over to access point mode
Dec 13 21:21:45 MAME82-P4WNP1 bash[178]: /home/pi/P4wnP1/boot/init_wifi.sh: line 38: fasle: command not found
Dec 13 21:21:46 MAME82-P4WNP1 dnsmasq[286]: started, version 2.76 DNS disabled
Dec 13 21:21:46 MAME82-P4WNP1 dnsmasq[286]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth 
Dec 13 21:21:46 MAME82-P4WNP1 dnsmasq-dhcp[286]: DHCP, IP range 172.24.0.2 -- 172.24.0.100, lease time 5m
Dec 13 21:21:46 MAME82-P4WNP1 dnsmasq-dhcp[286]: DHCP, sockets bound exclusively to interface wlan0
Dec 13 21:21:46 MAME82-P4WNP1 bash[178]: USB OTG off, going on with P4wnP1 boot
Dec 13 21:21:46 MAME82-P4WNP1 bash[178]: P4wnP1: ... USB gadget initialized
Dec 13 21:21:47 MAME82-P4WNP1 bash[178]: P4wnP1: Initializing Ethernet over USB...
Dec 13 21:21:47 MAME82-P4WNP1 bash[178]: Forwarding P4wnP1 SSH server to "your-ssh-server.com" ...
Dec 13 21:21:47 MAME82-P4WNP1 bash[178]:     P4wnP1 SSH will be reachable on localhost:8765 on this server
Dec 13 21:21:47 MAME82-P4WNP1 sudo[307]:     root : TTY=unknown ; PWD=/sys/kernel/config/usb_gadget/mame82gadget ; USER=root ; COMMAND=/usr/b
Dec 13 21:21:47 MAME82-P4WNP1 sudo[307]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 13 21:21:47 MAME82-P4WNP1 autossh[311]: port set to 0, monitoring disabled
Dec 13 21:21:47 MAME82-P4WNP1 sudo[307]: pam_unix(sudo:session): session closed for user root
Dec 13 21:21:47 MAME82-P4WNP1 autossh[314]: starting ssh (count 1)
Dec 13 21:21:47 MAME82-P4WNP1 autossh[314]: ssh child pid is 316
Dec 13 21:21:47 MAME82-P4WNP1 systemd[1]: Started P4wnP1 Startup Service.
Dec 13 21:21:47 MAME82-P4WNP1 bash[178]: Waiting for HID keyboard to be usable...
Dec 13 21:21:48 MAME82-P4WNP1 autossh[314]: ssh exited with error status 255; restarting ssh
Dec 13 21:21:48 MAME82-P4WNP1 autossh[314]: starting ssh (count 2)
Dec 13 21:21:48 MAME82-P4WNP1 autossh[314]: ssh child pid is 325
Dec 13 21:21:48 MAME82-P4WNP1 autossh[314]: ssh exited with error status 255; restarting ssh
Dec 13 21:21:48 MAME82-P4WNP1 autossh[314]: starting ssh (count 3)
Dec 13 21:21:48 MAME82-P4WNP1 autossh[314]: ssh child pid is 343
Dec 13 21:21:49 MAME82-P4WNP1 autossh[314]: ssh exited with error status 255; restarting ssh
Dec 13 21:21:49 MAME82-P4WNP1 autossh[314]: starting ssh (count 4)
Dec 13 21:21:49 MAME82-P4WNP1 autossh[314]: ssh child pid is 361
-- Logs begin at Thu 2016-11-03 17:16:43 UTC, end at Wed 2017-12-13 21:23:28 UTC. --
Dec 13 21:21:39 MAME82-P4WNP1 systemd[1]: Starting P4wnP1 Startup Service...
Dec 13 21:21:39 MAME82-P4WNP1 bash[178]: =================================== P4wnP1 startup ===========================================
Dec 13 21:21:39 MAME82-P4WNP1 bash[178]: P4wnP1: Init LED control...
Dec 13 21:21:40 MAME82-P4WNP1 bash[178]: P4wnP1: Loading config ...
Dec 13 21:21:41 MAME82-P4WNP1 bash[178]: P4wnP1: Initializing USB gadget ...
Dec 13 21:21:42 MAME82-P4WNP1 bash[178]: crw------- 1 root root 242, 0 Dec 13 21:21 /dev/hidg0
[... cut ...]
Dec 13 21:22:59 MAME82-P4WNP1 dnsmasq-dhcp[286]: 336405869 sent size:  4 option: 59 T2  4m22s
Dec 13 21:22:59 MAME82-P4WNP1 dnsmasq-dhcp[286]: 336405869 sent size:  4 option:  1 netmask  255.255.255.0
Dec 13 21:22:59 MAME82-P4WNP1 dnsmasq-dhcp[286]: 336405869 sent size:  4 option: 28 broadcast  172.24.0.255
Dec 13 21:22:59 MAME82-P4WNP1 dnsmasq-dhcp[286]: 336405869 sent size:  4 option: 12 hostname  kali

i hope its help, i have no check for the problem, the script and the project works not for me and other user, make a image for the PI Zero, than have nobody prblems with this!

Bye

Swiftb0y, 2017-12-14:
As I said `git pull && ./install.sh` won't work. You have to use `git pull --recursive https://github.com/mame82/P4wnP1 && cd P4wnP1 && ./install.sh`. The **--recursive** is important. 
You also didn't change any settings in the payload itself, which is why AutoSSH tries to start the whole time. You should disable that if you don't use it. 
And regarding the image. Apart from there being tedious work involved in creating a pre-done raspbian, there are licensing and copyright issues which is why we can't just distribute a custom version of raspbian. The script is already foolproof when you read the installation instructions as well as the pages on (the wiki)[http://p4wnp1.readthedocs.io/en/latest/Getting-Started-Subfolder/Installation/] carefully.

mr0Ot, 2017-12-14:
i use only this one: git pul && ./install.sh
today i will test this way:

git pull --recursive https://github.com/mame82/P4wnP1 && cd P4wnP1 && ./install.sh

i hope this works ;-)
Its o long way, i write you later.
Thanks for helping me.

Swiftb0y, 2017-12-14:
Save the log from the installation script and post it if it doesn't work again.

mr0Ot, 2017-12-14:
so im home and now i will check this install ;-)
First i install the Raspberry Pi Lite, works finde!

now i check what do you write:
git pull --recursive https://github.com/mame82/P4wnP1 && cd P4wnP1 && ./install.sh

An now we have the first problem:

pi@raspberrypi:~ $ git pull --recursive https://github.com/mame82/P4wnP1 && cd P4wnP1 && ./install.sh
fatal: Not a git repository (or any of the parent directories): .git
pi@raspberrypi:~ $

Ok now i will test it only with this one:

git clone --recursive https://github.com/mame82/P4wnP1 && cd P4wnP1 && ./install.sh

This way works the install run.
Can i use this way:

git clone pull --recursive https://github.com/mame82/P4wnP1 && cd P4wnP1 && ./install.sh

"git pull" works not for a fresh install, this works only when i install P4wnP1!
when it not works i test it with this steps:

1) Install raspberry p1 lite
2) sudo install -y git
3) cd /home/pi/
4) git clone  pull --recursive https://github.com/mame82/P4wnP1
5) cd P4wnP1
6) ./install.sh

i write you here.
bye

mame82, 2017-12-14:
Please use `clone` instead of `pull`.
The latter would pull down changes for an existing git from a remote. This only works if the P4wnP1 git already exists locally, but usn't the inteded update mechanism (new commits often change the installer script, which is meant to be run on a clean raspbian lite).

mr0Ot, 2017-12-14:
Sorry it works not!
the same problems, i hope in the future works this scrip.
i have no fun to testing the problems and nothink works with this P4wnP1, sorry is bullshit for me.

i come in 3 month back and i hope the children's diseases i eliminates.
i have no fun to be a alpha tester, sorry man and thanks for the help.

the idea with the iso should be seriously considered!
when someone has a working p4wnp1 times please the SDcard to iso and the whole with link upload!

mr0Ot, 2017-12-14:
Hi mame82, sorry for long text that i write, i test ist with the git pull, that works not look over my post you can see it:

git pull --recursive https://github.com/mame82/P4wnP1 && cd P4wnP1 && ./install.sh

WORKS NOT!
Look please here:

An now we have the first problem:

pi@raspberrypi:~ $ git pull --recursive https://github.com/mame82/P4wnP1 && cd P4wnP1 && ./install.sh
fatal: Not a git repository (or any of the parent directories): .git
pi@raspberrypi:~ $

thanks for all the help from you and Swiftb0y, i have no fun with your hard work, to many problems.
i come back in 3 month.

best regards

mame82, 2017-12-14:
Beside that the whole project is declared as 'experimental', while it is still undergoing major changes, there won't be a prebuilt image.

Feel free to ask somebody who's able to follow the install instructions for a (hopefully shrinked) image. Good luck

mame82, 2017-12-14:
Once more 'pull' doesn't work. 
Please stop polluting the issue for an OSX bug

mr0Ot, 2017-12-14:
yes sorry.
i write it from my mac, this mac use OSX and i hope that is the problem.
leperjulien write this:

-bash: /tmp/blink_count: Permission denied
There is no screen to be detached.

that was my problem.
now i test it with Ubuntu, Klai, Windows and Android, i use a ipad....
nothink works with this one!

ok now i write here not, is a problem on OSX right ;-)

leperjulien, 2017-12-20:
@mr0Ot 
WTF lol ?

It run perfectly on MacbookPro 2012, iMac 27" 2010, and my MacbookPro 2009

I can use duckyscript , send keys , i even created a payload that mount a fake usb store with a hidden binary of laZagne inside , grab all the password save them on p4wnp1 then reboot to another payload with another fake usb store with videos and stuff on this one

This is an amazing tool and i hope it gets better with time and help !

izeau, 2018-02-13:
I’ve looked into this as well, seems like the `open('/dev/hidg0','rb')` call in `boot_P4wnP1` hangs forever. I replaced it with a simple `sleep` for now (I’m guessing @leperjulien’s solution – hook into `onNetworkUp` instead of `onKeyboardUp` – has more or less the same effect). I’m not skilled enough to look further into how macOS handles HID keyboards but maybe we can add a timeout to the blocking call?

vinvinsim, 2021-09-26:
i have the same problem honestly the work around i used was when the assistant pops up it says that "you have pressed an unrecognized key on your keyboard press the key to the left of your shift button" (or something pretty close to that effect)..... at which point i unplugged my p4wnp1 and immidiatly plugged it back in to try to get it to "press the unrecognized key" again.... sure as shit first try  configured itself or should i say allowed the pi0 to be recognized as a keyboard + mouse.. i have replicated the results on seperste units with sucess all ive noticed is that sometimes it takes a few tries of plugging and unplugging to get it to "strike the key"