RoganDawes/P4wnP1 #170: DuckToolkit

iamalsaher, 2017-11-25:
**Is it possible to use payloads from the website https://ducktoolkit.com and use it on P4wnP1?** 

I feel it should work but I haven't succeeded in doing so. The provided ducky scripts work perfectly but the one I try to copy or make don't seem to work

Swiftb0y, 2017-11-25:
It should technically work. There must either be a bug in the generated DuckyScript or in MaMe's custom DuckyScriptEncoder. Could you please post your failing DuckyScript?

iamalsaher, 2017-11-27:
I will get back to you on this, maybe I am doing some mistake, I am actually new to this, so if it would be a genuine concern I will reply back, till then you can close the issue

closed 2018-04-07 by mame82