RoganDawes/P4wnP1 #171: Cannot SSH after Setup

davidyu96, 2017-11-26:
Hello:

After I ran through ./install.sh, I am unable to SSH into the pi using "pi@raspberrypi.local":
"Unable to open connection to raspberrypi.local. Host does not exist."

The strange thing is if I SSH into "raspberrypi", I get the "login as:" prompt. However, if I try to enter the user as "pi" and the password as "raspberry" (haven't changed the default password), it says "Access denied". 

The only way I can SSH into it is if I join the P4wnP1 wireless network and SSH using pi@172.24.0.1 (but that means I won't have internet connection). 

If I uninstall P4wnP1 and reformat the SD card with a fresh copy of Raspbian Stretch Lite, I can SSH into the pi normally. 

Any help would be greatly appreciated.

Swiftb0y, 2017-11-26:
the hostname gets changed by P4wnP1 do raspberrypi.local should not resolve to any host. I believe the new hostname would be MAME82-P4WNP1, but if you want to be sure just use pi@172.16.0.1 when SSHing over the virtual Ethernet adapter.

closed 2017-12-02 by davidyu96

cyb3r-n3rd, 2022-01-30:
Hey in  my P4WNP1 i have successfully completed  ./install.sh all the steps after that my pi gets disconnected, and still i was unable to ssh into my pi it is showing a hotspot named P4wnP1 but when i tried to nmap my network there is no device with port 22 open how could i ssh into my device