RoganDawes/P4wnP1 #172: hid backdoor : there is no screen to be detached

carmelo42, 2017-11-26:
Hello,
I have just created an microSD with raspbian + P4wnP1, but after rebooting with hid backdoor payload, when I connect to my pi0W I have this : 

```
-bash: /tmp/blink_count: Permission denied
There is no screen to be detached.
```

(instead of "Starting P4wnP1 server ..." screen)

Swiftb0y, 2017-11-26:
Make sure your installation up to date and if it is please provide the output of `sudo journalctl -u P4wnP1.service`. If it isn't, run `git pull && ./install` in your P4wnP1 directory.

carmelo42, 2017-11-26:
thanks for your answer !

`pi@MAME82-P4WNP1:~ $ sudo journalctl -u P4wnP1.service
-- Logs begin at Sun 2017-11-26 13:59:52 UTC, end at Sun 2017-11-26 14:04:07 UTC. --
Nov 26 13:59:57 MAME82-P4WNP1 systemd[1]: Starting P4wnP1 Startup Service...
Nov 26 13:59:57 MAME82-P4WNP1 bash[169]: =================================== P4wnP1 startup ===========================================
Nov 26 13:59:57 MAME82-P4WNP1 bash[169]: P4wnP1: Init LED control...
Nov 26 13:59:57 MAME82-P4WNP1 bash[169]: P4wnP1: Loading config ...
Nov 26 14:03:34 MAME82-P4WNP1 bash[169]: P4wnP1: Initializing USB gadget ...
Nov 26 14:03:35 MAME82-P4WNP1 bash[169]: crw------- 1 root root 242, 0 Nov 26 14:03 /dev/hidg0
Nov 26 14:03:35 MAME82-P4WNP1 bash[169]: crw------- 1 root root 242, 1 Nov 26 14:03 /dev/hidg1
Nov 26 14:03:35 MAME82-P4WNP1 bash[169]: crw------- 1 root root 242, 2 Nov 26 14:03 /dev/hidg2
Nov 26 14:03:35 MAME82-P4WNP1 bash[169]: crw------- 1 root root 242, 0 Nov 26 14:03 /dev/hidg0
Nov 26 14:03:35 MAME82-P4WNP1 bash[169]: crw------- 1 root root 242, 1 Nov 26 14:03 /dev/hidg1
Nov 26 14:03:35 MAME82-P4WNP1 bash[169]: crw------- 1 root root 242, 2 Nov 26 14:03 /dev/hidg2
Nov 26 14:03:35 MAME82-P4WNP1 bash[169]: P4wnP1: Checking for WiFi capabilities ...
Nov 26 14:03:36 MAME82-P4WNP1 bash[169]: P4wnP1: Seems WiFi module is present !
Nov 26 14:03:37 MAME82-P4WNP1 sudo[252]:     root : TTY=unknown ; PWD=/sys/kernel/config/usb_gadget/mame82gadget ; USER=root ; COMMAND=/sbin/ifconfig wl
Nov 26 14:03:37 MAME82-P4WNP1 sudo[252]: pam_unix(sudo:session): session opened for user root by (uid=0)
Nov 26 14:03:37 MAME82-P4WNP1 sudo[252]: pam_unix(sudo:session): session closed for user root
Nov 26 14:03:37 MAME82-P4WNP1 bash[169]: Try to find WiFi AC329
Nov 26 14:03:37 MAME82-P4WNP1 sudo[262]:     root : TTY=unknown ; PWD=/sys/kernel/config/usb_gadget/mame82gadget ; USER=root ; COMMAND=/sbin/iwlist wlan
Nov 26 14:03:37 MAME82-P4WNP1 sudo[262]: pam_unix(sudo:session): session opened for user root by (uid=0)
Nov 26 14:03:38 MAME82-P4WNP1 sudo[262]: pam_unix(sudo:session): session closed for user root
Nov 26 14:03:38 MAME82-P4WNP1 bash[169]: Network AC329 found
Nov 26 14:03:38 MAME82-P4WNP1 bash[169]: ... creating config
Nov 26 14:03:38 MAME82-P4WNP1 sudo[281]:     root : TTY=unknown ; PWD=/sys/kernel/config/usb_gadget/mame82gadget ; USER=root ; COMMAND=/bin/bash -c cat
Nov 26 14:03:38 MAME82-P4WNP1 sudo[281]: pam_unix(sudo:session): session opened for user root by (uid=0)
Nov 26 14:03:38 MAME82-P4WNP1 sudo[281]: pam_unix(sudo:session): session closed for user root
Nov 26 14:03:38 MAME82-P4WNP1 sudo[293]:     root : TTY=unknown ; PWD=/sys/kernel/config/usb_gadget/mame82gadget ; USER=root ; COMMAND=/bin/bash -c cat
Nov 26 14:03:38 MAME82-P4WNP1 sudo[293]: pam_unix(sudo:session): session opened for user root by (uid=0)
Nov 26 14:03:39 MAME82-P4WNP1 sudo[293]: pam_unix(sudo:session): session closed for user root
Nov 26 14:03:39 MAME82-P4WNP1 bash[169]: ... connecting ...
Nov 26 14:03:39 MAME82-P4WNP1 sudo[302]:     root : TTY=unknown ; PWD=/sys/kernel/config/usb_gadget/mame82gadget ; USER=root ; COMMAND=/sbin/wpa_supplic
Nov 26 14:03:39 MAME82-P4WNP1 sudo[302]: pam_unix(sudo:session): session opened for user root by (uid=0)
Nov 26 14:03:39 MAME82-P4WNP1 bash[169]: Successfully initialized wpa_supplicant
Nov 26 14:03:39 MAME82-P4WNP1 sudo[302]: pam_unix(sudo:session): session closed for user root
Nov 26 14:03:40 MAME82-P4WNP1 sudo[315]:     root : TTY=unknown ; PWD=/sys/kernel/config/usb_gadget/mame82gadget ; USER=root ; COMMAND=/sbin/dhclient -4
Nov 26 14:03:40 MAME82-P4WNP1 sudo[315]: pam_unix(sudo:session): session opened for user root by (uid=0)
Nov 26 14:03:40 MAME82-P4WNP1 dhclient[319]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
Nov 26 14:03:40 MAME82-P4WNP1 sudo[315]: pam_unix(sudo:session): session closed for user root
Nov 26 14:03:40 MAME82-P4WNP1 bash[169]: USB OTG off, going on with P4wnP1 boot
Nov 26 14:03:40 MAME82-P4WNP1 bash[169]: P4wnP1: ... USB gadget initialized
Nov 26 14:03:40 MAME82-P4WNP1 bash[169]: P4wnP1: Initializing Ethernet over USB...
Nov 26 14:03:40 MAME82-P4WNP1 systemd[1]: Started P4wnP1 Startup Service.
Nov 26 14:03:40 MAME82-P4WNP1 bash[169]: Waiting for HID keyboard to be usable...
Nov 26 14:03:41 MAME82-P4WNP1 bash[169]: 0
Nov 26 14:03:46 MAME82-P4WNP1 dhclient[324]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 20
Nov 26 14:03:46 MAME82-P4WNP1 dhclient[324]: DHCPREQUEST of 192.168.1.111 on wlan0 to 255.255.255.255 port 67
Nov 26 14:03:46 MAME82-P4WNP1 dhclient[324]: DHCPOFFER of 192.168.1.111 from 192.168.1.10
Nov 26 14:03:47 MAME82-P4WNP1 dhclient[324]: DHCPACK of 192.168.1.111 from 192.168.1.10
Nov 26 14:03:48 MAME82-P4WNP1 dhclient[324]: bound to 192.168.1.111 -- renewal in 34593 seconds.
`

I have done 👍 
sudo apt-get update

and git pull && ./install in the P4wnP1 dir already

Feared-Penguin, 2017-11-30:
I'm having the same issue, I've got a fresh version of raspbian updated and upgraded, and p4wn pi was cloned an hour ago.

any ideas?

mame82, 2018-04-07:
Please try the ready-to-go image provided here https://github.com/mame82/P4wnP1/releases

sircosec, 2018-07-28:
I've tried the both the release version and the pre-built alpha and I get the same issue from both. Is there any other feedback I can provide?

Swiftb0y, 2018-07-28:
Sometimes I have the same issue. My workaround is that I just start the server manually from hidtools/backoor/P4wnpy.py

sircosec, 2018-07-28:
thanks @Swiftb0y that helped. i didn't know where that script was or what it was called. let me know if there's anything I can do to support fixing the issue in the image.

Moe-New, 2018-11-23:
Which payload do you use?
You can check the payload file to see how it works.
If it is hid_backdoor,you can run
 `sudo screen -dmS hidsrv bash -c "cd $wdir/hidtools/backdoor; python P4wnP1.py"`

GermanNoob, 2019-03-31:
Just to be sure that no one else is as dumb as I was:

Check your USB cable if it supports DATA!!! I grabbed one which just delivered power... 😳
It took me hours to realize this... 🤪

closed 2023-05-14 by carmelo42