RoganDawes/P4wnP1 #174: duckhid strange things , please help

beboxos, 2017-11-27:
i have attached a physical button to my pi , then i want to run a duckscrpit when press on it 
all the way , button etc is ok but i fail sending keys 

on terminal ssh if type for example 
cat /home/pi/P4wnP1/DuckyScripts/HelloWorld.duck | duckhid
work like a charm , run notepad and type hello the world

if i make a hello.sh
#!/bin/bash
cat /home/pi/P4wnP1/DuckyScripts/HelloWorld.duck | duckhid

don't work i have this error ..

pi@MAME82-P4WNP1:~/P4wnP1 $ sh hello.sh
hello.sh: 2: hello.sh: duckhid: not found
pi@MAME82-P4WNP1:~/P4wnP1 $

by the way wanted to try call this to python but don't know how :'( 
thinked use an in my python script
import os
 os.system('sh /home/pi/hello.sh')

hahnstep, 2017-11-27:
duckhid is not an cmd is a function inside P4wnP1/boot/init_hid_keyboard.sh 
You can use python /home/pi/P4wnP1/duckencoder/duckencoder.py to send keys.

beboxos, 2017-11-27:
thank you hahnstep 

work fine now with 

os.system('cat /home/pi/P4wnP1/DuckyScripts/HelloWorld.duck | cat | python /home/pi/P4wnP1/duckencoder/duckencoder.py -l fr -p | python /home/pi/P4wnP1/hidtools/transhid.py')

(yes looonng command, but it's working :) )

closed 2017-11-27 by beboxos

Swiftb0y, 2017-11-27:
I would suggest looking at hidtools/backdoor/P4wnP1.py to figure out how Sendkeys is implemented.

mame82, 2017-11-27:
DuckEncoder.py is implemented as a Class and could directly be imported to a python script. I suggest following the advice of @Swiftb0y and to take a look into P4wnP1.py as I haven't documented this functionality