RoganDawes/P4wnP1 #176: When rebooting, setup.cfg reset

MrSbarre, 2017-12-02:
Hi ! 
I have a big situation, every time I reboot the raspberry, the setup.cfg file just reset himself. So when I modify it to use the HID backdoor for exemple, it just don't want to keep the change after the reboot.
I did Ctrl+X and yes, when I return to the file after I modified it, it is still modified, but like I said, it just reset when I power up the raspberry.

Thanks in advance !

Swiftb0y, 2017-12-02:
Usually the setup.cfg doesn't get reset, but every relevant option gets overridden by the selected payload. The setup.cfg is only there to provide defaults in case there is nothing defined in payload itself.

MrSbarre, 2017-12-02:
Yes, but I'm talking about the PAYLOAD options at the end, when I deselecte
the original (network) and run the HID backdoor for exemple, at boot, it
reput the payload at network.

IronPhreak, 2018-01-05:
Hi,

Just to confirm, are you modifying the Payload file, HID_backdoor.... or the setup.cfg file?

As stated by Swiftb0y and in my experience, setup.cfg arguments will get overwritten by the payload you're loading however it doesn't change the setup.cfg file.

mame82, 2018-04-07:
Closed ... missing feedback

closed 2018-04-07 by mame82