RoganDawes/P4wnP1 #177: Cannot Connect to the Pi Zero W with USB and SSH after Installing P4wnP1

davidyu96, 2017-12-02:
My ISP blocks the use of SSH over my wireless network, so in order for me to interface with the Pi, I had to emulate ethernet over usb by:
1. Appending "dtoverlay=dwc2" to config.txt
2. Adding "modules-load=dwc2,g_ether" after "rootwait" in cmdline.txt
3. Creating a file called "ssh" in /boot
After doing these steps, I can ssh into pi@raspberrypi.local via PuTTY (Pi is running Raspbian Stretch).

However, after installing P4wnP1, I am unable to ssh into pi@172.16.0.1 and pi@172.16.0.1.local. Is there a way to continue connecting to the Pi with USB and SSH with P4wnP1 installed, or is this impossible?

Thanks,
David

Swiftb0y, 2017-12-02:
Could you please provide the output of `sudo journalctl -u P4wnP1.service` and `ifconfig`

Anonyvius, 2018-01-14:
wich image does you use? Jessie lite or stretch lite? because for stretch you would need:

`country=DE  #omit if US`
`ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev`
`update_config=1`
`network={`
       `ssid="wlan-bezeichnung"`
       `psk="passwort"`
       `key_mgmt=WPA-PSK`
`}`

mame82, 2018-04-07:
Closed due to missing feedback

closed 2018-04-07 by mame82