RoganDawes/P4wnP1 #180: -bash: /tmp/blink_count: Permission denied

mr0Ot, 2017-12-12:
Hi a have a question.
when i will start the: "PAYLOAD-hid _backdoor.txt "

-bash: /tmp/blink_count: Permission denied 
There is no screen to be detached.﻿

Please can you tell me waht i make wrong?
i understand not what i must do.
it will make premission, ok, what must i do?

THX for the help

Swiftb0y, 2017-12-12:
I correctly don't have an up to date pull, but I think this is just a minor inconvenience due to a broken udev configuration. This won't break any significant features of P4wnP1.
The `There is no screen to be detached` is an due to an issue of the client and screen. You can use `screen` to get a screen session that persists even if your SSH connection breaks and `python hidtools/backdoor/P4wnP1.py` in the P4wnP1 directory and to start the server manually. This is only a workaround but I suspect the main issues you are encountering will be fixed once the new Wireless features are stable and merged into the master branch.

mr0Ot, 2017-12-12:
I hope the fix come ;-)
I test tomorrow a new install form this.
thank you for the help.

best regards

closed 2018-04-07 by mame82