RoganDawes/P4wnP1 #181: [feature request]Keylogger

amohacker, 2017-12-12:
Hey, I have an idea for a possable feature.
You could make a keylogger that saves all keystrokes of the attached PC or maybe make it so that it installs a keylogger that saves all keystrokes and when the P4wnP1 is attached again it saves it on it.
Hope you like the idea.

PS: love what you're doing

mr0Ot, 2017-12-12:
Look here:
https://github.com/mame82/P4wnP1/issues/138

closed 2017-12-12 by amohacker

mr0Ot, 2017-12-12:
Sorry i found new infos from the web.
The P4wnP1 habe the script DuckyScripts.
The DuckyScripts works with a keylogger, i hope i tell not bullshit.
i found a video, look here: 

https://www.youtube.com/watch?v=_iuHZUP_qsQ

He tells from a script that is this one:

pykeylogger

And he tell that it works.
i hope for that i tell you good thinks.