RoganDawes/P4wnP1 #182: unable to use setup.cfg to modify hostapd setting.

triplettjon, 2017-12-18:
setup.cfg modify access point setting is not adjusting hostapd.cfg keeping the same ap default name. 

function start_wifi_accesspoint()
{
	generate_hostapd_conf

	hostapd -d /tmp/hostapd.conf > /tmp/hostapd.log &

	# configure interface
	ifconfig wlan0 $WIFI_ACCESSPOINT_IP netmask $WIFI_ACCESSPOINT_NETMASK

	# start DHCP server (second instance if USB over Etherne is in use)
	generate_dnsmasq_wifi_conf
	dnsmasq -C /tmp/dnsmasq_wifi.conf

mame82, 2017-12-18:
Could you please clarify

triplettjon, 2017-12-18:
Sorry Marcus for being so unclear. My intent is not to waste any of your
time as I know your busy. I just wanted to share what I noticed.

I'm going to rebuild and make sure I didn't screw some thing up and do some
further testing.

 In my case when I modify the setup.cfg to a different ssid and password.
it defaults back to the p4wnp1 and default password. I checked the hostapd
and it was the default ssid "of course". now after running the device and
doing a few restarts it finally used my new ssid from the setup.cfg. after
powering off and back on it would go right back to the default.  "which
confused me"

I was using both otg and straight powered for both tests. to see if it was
some thing strange about its selection of the scripts but I didn't notice
any changes.

anyways I hope this explained what I was seeing.  I will be working more to
see if i can't ad some pie to phone options. I see this as an amazing cheap
project and hardware with a great pen testing kit.

thanks

Jonathon

Swiftb0y, 2017-12-18:
Seems like you are over analyzing the problem: #156 is probably the solution.

triplettjon, 2017-12-18:
most likely im over analyzing :) for some reason I was assuming that the setup.cfg would override the settings on the payloads but to be honest I dont see it necessary.

Swiftb0y, 2017-12-18:
Well, the setup.cfg is just there to provide defaults. So the settings that are needed by the Payload are set at the top of the payload itself. This is only documented in the wiki which gets read by far to less people.

closed 2017-12-18 by triplettjon