RoganDawes/P4wnP1 #183: Does it work on smartphones

amohacker, 2017-12-18:
Is it posseble to use it to run duckyscripts on a smartphone?

hahnstep, 2017-12-18:
yes, one example here  
https://github.com/anbud/DroidDucky

amohacker, 2017-12-18:
What i meant was can you use the p4wnp1 to execute duckyscripts to(?) phones

hahnstep, 2017-12-18:
you mean a payload like this ?
https://forums.hak5.org/topic/28165-payload-android-brute-force-4-digit-pin/

amohacker, 2017-12-18:
Yeah

Swiftb0y, 2017-12-18:
Im pretty sure it does. Even relative mouse movements should work on android. Wasn't able to test it your myself though. You just need an extra OTG adapter.

hahnstep, 2017-12-19:
I tested with an Nexus 5 and it dont work, the USB OTG Mode does prevent it.

mame82, 2017-12-19:
It works against phones. I've done a bunch of experiments with emulating digitizer boards (for absolute cursor positioning), but the phones differ too much. Anyway, MouseScript works very well against Android (except absolute positioning).

Attach an OTG adapter to the phone, connect P4wnP1 like every other USB peripheral and you should be good to go emulating keyboard/mouse.

Tested on Android (7.0), CM 14 against several Samsung devices.

Hid_mouse payload is a good starting point (comments in payload)

mame82, 2017-12-19:
Sorry, comments are in the MouseScript

https://github.com/mame82/P4wnP1/blob/master/MouseScripts/test.mouse

The payload has yo be modified to disable the keyboard trigger (would be hard to press NUMLOCK frequently  on the Smartphone ;-))

amohacker, 2017-12-19:
The backdoor didn't trigger on Android 5.1.1 but ill try again maybe i did something wrong

mame82, 2017-12-19:
The backdoor is Windows only. The client side code is PowerShell/Net based. 
Although the README is a bit outdated, there're enough details on the inner workings of the backdoor, which should clarify that this is Windows only.

P4wnP1 provides the potential to target those devices anyway. Example killchain:
- prepare a backdoored apk (f.e. meterpreter) via http over WiFi
- have a payload handler running on the same WiFi or external server
- prepare a keyboard script, which opens a browser, downloads the apk and confirms the respective dialogs via keystroke injection after joining the WiFi
- conmect P4wnP1 via OTG ro the Android to carry out the keystroke attack
- receive a shell

Be aware of the fact, that I don't provide such payloads for two reasons:
1. My disclaimer
2. They're very device specific when it comes to UI interaction (especially if this should be done using mouse). But there're a few shortcuts common amongst different android versions (including the ones to open a browser or send emails).

Summery:
P4wnP1 could be used against Android if the payload is well prepared, the hid_backdoor is Windows only

amohacker, 2017-12-19:
I got that the backdoor doesn't work but you should still be able to launch duckyscript's through it right?

mame82, 2017-12-19:
Yes, the console interface could be used for this ... Same goes for MouseScript with `SendMouseScript`

closed 2018-04-07 by mame82

bogiton, 2018-04-16:
What about its capability as a network adapter? Could that work under an Android device to create a network between the pi and the phone?